eblot/pyftdi #131: SMBUS using pyftdi

aplak1, 2019-06-14:
Is it possible to implement SMBUS protocol using the functions offered by the pyftdi library?
I see there is a group of functions for I2C but I would like to know if anyone has ever tried to implement SMBUS using these functions. In particular I need to implement the "Block Write Block Read Proc Call" function.
Thanks!

eblot, 2019-12-02:
I do not know the SMBUS protocol, but from https://www.kernel.org/doc/Documentation/i2c/smbus-protocol it looks like it is an I2C transaction made of write-then-read. I guess this can be implemented with PyFtdi, with something like:

    # emit a start, send address, write data, do not release the bus
    i2cport.write(self, out, relax=False)
    # send address, do not emit a start, read data, release the bus
    i2cport.read(self, len, start=False)

eblot, 2020-02-07:
No feedback. Feel free to reopen w/ new info

closed 2020-02-07 by eblot