eblot/pyftdi #133: GPIO Addressable Pins?

dua564, 2019-06-21:
Hi eblot,

Was wondering, I'm using another FT232H solely as I2C Controller with 11 GPIO pins.
Does the GPIO library only allow for 8 GPIO pins? My GPIO pins are D5-D7, C0-C7. How does I address these in the direction field in self.gpio.open_from_url(url=url, direction=direction)

Can direction be set to 0xFFF0 or is this limited to only 8 bits? 

Thanks!

Nik

eblot, 2019-06-23:
I2C + GPIO is not yet supported.

However, there is a PR (#99) to address this topic. It has not been merged as I do not have time to write unit tests for now.

dua564, 2019-06-26:
Hi eblot,

Thanks. I was able to get this to work!

-Nik

eblot, 2019-08-29:
I have just committed a new version for I2C + GPIO support, based on latest PyFtdi version.

Checkout out `i2c_gpio` branch

eblot, 2019-08-29:
Now available in [v0.30.0](https://github.com/eblot/pyftdi/releases/tag/v0.30.0).

closed 2019-08-29 by eblot