eblot/pyftdi #134: AttributeError: function 'usb_find_busses' not found

sbouge13a, 2019-07-08:
>>> ftdi_i2c.configure('ftdi://ftdi:232h/1')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\steph\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyftdi\i2c.py", line 363, in configure
    frequency=frequency, **kwargs)
  File "C:\Users\steph\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyftdi\ftdi.py", line 471, in open_mpsse_from_url
    vendor, product, index, serial, interface = self.get_identifiers(url)
  File "C:\Users\steph\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyftdi\ftdi.py", line 309, in get_identifiers
    cls.SCHEME, cls.VENDOR_IDS, cls.PRODUCT_IDS, cls.DEFAULT_VENDOR)
  File "C:\Users\steph\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyftdi\usbtools.py", line 342, in parse_url
    devices = devclass.find_all(vps)
  File "C:\Users\steph\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyftdi\ftdi.py", line 390, in find_all
    return UsbTools.find_all(vps, nocache)
  File "C:\Users\steph\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyftdi\usbtools.py", line 58, in find_all
    devs.update(UsbTools._find_devices(v, p, nocache))
  File "C:\Users\steph\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyftdi\usbtools.py", line 242, in _find_devices
    for dev in backend.enumerate_devices():
  File "C:\Users\steph\AppData\Local\Programs\Python\Python37-32\lib\site-packages\usb\backend\libusb0.py", line 440, in enumerate_devices
    _check(_lib.usb_find_busses())
  File "C:\Users\steph\AppData\Local\Programs\Python\Python37-32\lib\ctypes\__init__.py", line 369, in __getattr__
    func = self.__getitem__(name)
  File "C:\Users\steph\AppData\Local\Programs\Python\Python37-32\lib\ctypes\__init__.py", line 374, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: function 'usb_find_busses' not found

I can't get it to work and I keep receiving this error

eblot, 2019-07-09:
It looks like a PyUSB issue, not a PyFTDI issue.

Something maybe unexpected is the USB backend: libusb0 refers to a very old USB backend - although I do not know much about Windows. Maybe the usb_find_busses is not supported with this old backend? Which versions of PyFtdi, PyUSB and libusb are you using?

sbouge13a, 2019-07-09:
I'm using pyusb 1.0.2, pyftdi 0.29.4 and libusb 1.0.22b4

sbouge13a, 2019-07-09:
I fixed the problem. I needed to add my path to the environment variables.

closed 2019-07-12 by eblot