eblot/pyftdi #136: Open device by bus number and address

RyanHope, 2019-07-26:
I have a bunch of fake ftdi chips that all have the same manu/product/serial info, When I used pyftdi's methods of listing devices by vendor id I only list one of the adapters if I have multiple connected to the same machine. The only way to identify them is by their bus number and address. I couldn't think of a good way to create a new open method that takes bus number/address so instead I created a new open method that accepts a device directly as returned by usb.core.find. This seems to be working fine. I just extended the Ftdi class in my project but I could make a proper PR if interested.

eblot, 2019-08-01:
You can add a PR. I'm not sure it can be included into the project, but it could help other users with EEPROM-less boards. Thanks.

RyanHope, 2019-08-01:
I'm not sure that this issue is specific for fake chips though. Your code
should still be able to list and select from multiple devices if multiple
are present.

eblot, 2019-08-01:
It is not related to fake chips but to the absence of EEPROM (or properly flashed EEPROM).

However, the code does work with several identical chips connected:

Example:
`url=ftdi:///?`:

     Available interfaces:
      ftdi://ftdi:232:FTH19HLV/1    (TTL-232R-3V3-AJ)
      ftdi://ftdi:232:FTH1GMPE/1    (TTL-232R-3V3-AJ)
      ftdi://ftdi:232h:FTVOLDVA/1   (C232HD-DDHSP-0)

- first TTL-232R-3V3-AJ can be open with either `url=ftdi://ftdi:232:FTH19HLV/1` (by serial number) or `url=ftdi://ftdi:232:1/1` (by index)
- second TTL-232R-3V3-AJ can be open with either `url=ftdi://ftdi:232:FTH1GMPE/1` (by serial number) or `url=ftdi://ftdi:232:2/1` (by index)

The trouble is to identify which device is actually connected, I'm not sure the bus/address can help as the USB bus may be re-enumerated and bus/address keep changing.

RyanHope, 2019-08-01:
That's a bad example, the devices are not identical as they have different
serial numbers.

eblot, 2019-08-01:
Ok, I got it: I never tested a case with identical chips and no (or identical) serial number. It should definitely be fixed.

eblot, 2019-08-01:
I will introduce bus and address explicit syntax, and use `open_from_device`:

    Available interfaces:
      ftdi://ftdi:232h:FTH19HLV/1   (TTL-232R-3V3-AJ)
      ftdi://ftdi:232h:FTH1GMPE/1   (TTL-232R-3V3-AJ)
      ftdi://ftdi:232h:14:2c/1
      ftdi://ftdi:232h:14:2e/1

4 distinct FTDI devices in the above example, two with S/N, two other ones w/o.

eblot, 2019-08-01:
See usb_bus_address branch (WIP)

zhenqlin, 2019-08-20:
I encountered similar issue. 
I have FT2232H device with same serial number. when I use spi.configure('ftdi://::/1'), sometime connect to A channel sometime connect to B and spi.configure('ftdi://::/2') doesn't work. 
How to fix this?

eblot, 2019-08-21:
@zhenqlin I'm not sure to understand your message:

Do you have two (or more) FTDI devices w/ no serial numbers?
 * ftdi://:/1 means "first enumerated FTDI device, port A"
 * ftdi://:/2 means "first enumerated FTDI device, port B"

I do not see how the host could use A or B channels randomly, as channel selection is not related to device selection. It can change which FTDI device it uses each time your Python application is run, but .../1 always means "channel A" and .../2 means "channel B"

Please give a try to the `usb_bus_address` branch and use the new `bus:address` specifiers to select a specific device, ''i.e.'': `ftdi://::bus:address/1` and `ftdi://::bus:address/2`.

zhenqlin, 2019-08-21:
Thanks for reply. Doesn't look like working.
here's what I try.
code:
device = usb.core.find(find_all=True)
for d in device:
	e ={}
	e['vid'] = hex(d.idVendor)
	e['pid'] = hex(d.idProduct)
	e['bus'] = d.bus
	e['address'] = d.address
	e['serial']  = usb.util.get_string(d, d.iSerialNumber)
	print(e)
output:
{'vid': '0x6cb', 'pid': '0x91', 'bus': 0, 'address': 1, 'serial': 'FTA8QAPB'}
{'vid': '0x6cb', 'pid': '0x91', 'bus': 0, 'address': 2, 'serial': 'FTA8QAPB'}

Code:
ctrl = SpiController(cs_count=1, turbo=True)
Ftdi.add_custom_vendor(0x6cb)
Ftdi.add_custom_product(0x6cb, 0x91)
spi = SpiController(cs_count = 1)
spi.configure('ftdi://::0:1/1')

slave = spi.get_port(cs=0, freq=12E6, mode=0)
gpio = spi.get_gpio()
b_io=GpioController()
b_io.open_from_url('ftdi://::0:2/2', direction=0)

output:
Traceback (most recent call last):
  File "C:\temp\Rev_05_dev\usbtools.py", line 408, in parse_url
    desc, _ = candidates[idx]
IndexError: list index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test.py", line 27, in <module>
    b_io.open_from_url('ftdi://::0:2/2', direction=0)
  File "C:\temp\Rev_05_dev\gpio.py", line 79, in configure
    ftdi.open_bitbang_from_url(url, direction=direction, **kwargs)
  File "C:\temp\Rev_05_dev\ftdi.py", line 592, in open_bitbang_from_url
    devdesc, interface = self.get_identifiers(url)
  File "C:\temp\Rev_05_dev\ftdi.py", line 308, in get_identifiers
    cls.DEFAULT_VENDOR)
  File "C:\temp\Rev_05_dev\usbtools.py", line 412, in parse_url
    urlstr)
usbtools.UsbToolsError: No USB device matches URL ftdi://::0:2/2

zhenqlin, 2019-08-22:
I think usbtools.py find_all function use set for devs, that way sometime connect to channel A sometime B for my device. 
I hardcore ftdi.py bypass usbtools.py for open_mpsse_from_url and open_bitbang_from_url function looks like OK to connect now. Need continue move on to see if any other issues.

		addr = url.split('/')
		interface =  int(url.split('/')[-1])
		addr = addr[2].split(':')[-1]
		devices = usb.core.find(find_all=True)
		device = None
		for d in devices:
			if d.address == int(addr):
				device = d
		# devdesc, interface = self.get_identifiers(url)
		# device = UsbTools.get_device(devdesc)

eblot, 2019-08-22:
> Thanks for reply. Doesn't look like working.
> here's what I try.
> ...

Please try this, change to top level dir, and query the FTDI device list

     cd pyftdi
     PYTHONPATH=. python3 pyftdi/serialext/tests/pyterm.py -p ftdi:///?

what does that gives you?

I'm not sure to understand your HW setup. How many FTDI devices are plugged in your host?

eblot, 2019-08-22:
> I think usbtools.py find_all function use set for devs, that way sometime connect to channel A sometime B for my device.

`usbbools.py` manages the devices (*i.e.* FTDI devices), not the interface on a device (*i.e.* 1 channel for FTDI232, 2 channels for a single FTDI2232, 4 channels for a single FTDI4232), so I do not get why channels would get mixed. `usbtools.py` selects a device, `ftdi.py` manages the channels.

zhenqlin, 2019-08-22:
>  PYTHONPATH=. python3 pyftdi/serialext/tests/pyterm.py -p ftdi:///?

This test script is for Linux? I'm in windows,  
from os import linesep, name as osname, stat , uname
doesn't work, I took out uname and output is. 
No USB-Serial device has been detected.

I have only one device FT2232H, which has A and B two usb virtual port? and the different between them is the "address" .
output:
{'vid': '0x6cb', 'pid': '0x91', 'bus': 0, 'address': 1, 'serial': 'FTA8QAPB'}
{'vid': '0x6cb', 'pid': '0x91', 'bus': 0, 'address': 2, 'serial': 'FTA8QAPB'}

zhenqlin, 2019-08-22:
I plug my device into Linux pc. I see what you mean now. 
For Linux sees one device(same address) content two interface. 
But when I use Zadig install driver in windows 10, is two device (different address), one interface for each.

eblot, 2019-08-23:
> This test script is for Linux?
Any unix/posix OS (Linux, FreeBSD, macOS, ...)

> I'm in windows,
Ok... it explains nearly everything them. Windows is a gigantic pile of crap in general, and especially when it comes to USB.

For some reason - as you've seen since - Windows (and/or Zadig?) thinks it is great to present a multiple interface USB device as several devices. This is a non sense (try to reset or unplug one w/o the other...). I'm not surprise this confuses PyFtdi.

I remember to check how PyFtdi behaves on Windows - although it is not supported on Windows - and eventually find a setup with Zadig where I was able to see the FTDI device as, at once:

  1. a single device with two interface, AND
  2. two devices with a single interface

I do not remember how to reach this state though - I simply cannot cope with Windows, it gets on my nerves as soon as I see the logo...

Using the first reported device solved the issue, and you do no need the `usb_bus_address` branch at all to make it work - as you have a serial number on your device.

I'll try to have a look at this M$ mess, but I cannot tell when. Meanwhile, if you can find the proper Zadig setup to make the device appears as it really is (one device, two interfaces), your problem should be solved.

eblot, 2019-08-23:
In Zadig, if you check/uncheck the `Options/Ignore hubs or composite parents` menu item:

[image]

the FTDI2232 which appears a two unique devices:

[image]

now appears as a single device. 

[image]

Maybe this is the way to go. 

I cannot help much more, as whether or not I install the Zadig driver, `pyusb` seems unable to find `libusb`, and I already lost too much time with this ugly OS.

zhenqlin, 2019-08-24:
I know Linux developer always hate win 😊
Just like me, I want my software support both win and Linux, everything works on win but nothing works Linux.. 
OK! I change to Linux now, and lost a lot of time finally make the udev works..
If I have Linux questions in the future you can help, right??

eblot, 2019-08-25:
> OK! I change to Linux now, and lost a lot of time finally make the udev works..

Anything different than [installation.html#debian-ubuntu-linux](https://eblot.github.io/pyftdi/installation.html#debian-ubuntu-linux) ?

> If I have Linux questions in the future you can help, right??

I'll try - when I get some spare time :-)

BTW, did you try to setup the composite parent w/ Zadig?

RyanHope, 2019-08-25:
I develop on both win and linux... I will try and get some windows testing
in.

eblot, 2019-08-26:
Thanks Ryan. I'm unable to test it under Windows as pyusb fails to find Zadig's libusb for some reason.

eblot, 2019-08-26:
Ok, I was failing to select "libusb-win32" as the driver to install. Anyway, selecting the composite parent vs. an individual interface gives the expected result:

    C:\Users\eblot>pip3 install pyftdi
    Collecting pyftdi
    Requirement already satisfied: pyusb>=1.0.0 (1.0.2)
    Requirement already satisfied: pyserial>=3.0 (3.4)
    Installing collected packages: pyftdi
      Running setup.py install for pyftdi ... done
    Successfully installed pyftdi-0.29.6

    C:\Users\eblot>python
    Python 3.7.3 (v3.7.3:ef4ec6ed12, Mar 25 2019, 22:22:05) 
    Type "help", "copyright", "credits" or "license" for more information.
    >>> from pyftdi import ftdi
    >>> f = ftdi.Ftdi()
    >>> f.open_from_url('ftdi:///?')
    Available interfaces:
      ftdi://ftdi:2232/1   (Dual RS232-HS)
      ftdi://ftdi:2232/2   (Dual RS232-HS)
    
    Please specify the USB device
    
Note that is using the "master" branch, not the `usb_bus_address`. I have not tried to communicate over SPI, but there should be not channel permutation using this approach: 

 * install the driver for the composite parent, **not** for the individual interfaces of an FTDI device.

I should add this to the documentation. Documenting for Windows, my dreams came true :-)

eblot, 2019-08-26:
Using `usb_bus_address` branch:

    >>> f.open_from_url('ftdi:///?')
    Available interfaces:
      ftdi://ftdi:2232:0:1/1   (Dual RS232-HS)
      ftdi://ftdi:2232:0:1/2   (Dual RS232-HS)

Same bus, same address. At least...

eblot, 2019-08-26:
> I should add this to the documentation. Documenting for Windows, my dreams came true :-)

Done as d0b6366d

eblot, 2019-08-29:
@RyanHope the `bus:address` selection is now implemented in the `master` branch
@zhenqlin 
  1. I think the Windows problem is addressed with the now-documented Zadig configuration process
  2. If you have some issue with I2C, please open a new, dedicated ticket. Be sure to test the v0.30.0 version first, released today.

closed 2019-08-29 by eblot