eblot/pyftdi #137: License clarification

eclipseo, 2019-07-26:
In pyftdi/doc/licenses.rst you claim the program is licensed under MIT but actually provide the text of the BSD 3-Clause license. Moreover, the headers of py source file contains BSD 3-Clause license:

BSD 3-clause "New" or "Revised" License
---------------------------------------
pyftdi-0.29.2/pyftdi/__init__.py
pyftdi-0.29.2/pyftdi/bin/i2cscan.py
pyftdi-0.29.2/pyftdi/bits.py
pyftdi-0.29.2/pyftdi/gpio.py
pyftdi-0.29.2/pyftdi/i2c.py
pyftdi-0.29.2/pyftdi/misc.py
pyftdi-0.29.2/pyftdi/serialext/__init__.py
pyftdi-0.29.2/pyftdi/serialext/logger.py
pyftdi-0.29.2/pyftdi/serialext/protocol_ftdi.py
pyftdi-0.29.2/pyftdi/serialext/protocol_unix.py
pyftdi-0.29.2/pyftdi/serialext/tests/pyterm.py
pyftdi-0.29.2/pyftdi/serialext/tests/term.py
pyftdi-0.29.2/pyftdi/spi.py
pyftdi-0.29.2/pyftdi/tests/bits.py
pyftdi-0.29.2/pyftdi/tests/ftdi.py
pyftdi-0.29.2/pyftdi/tests/gpio.py
pyftdi-0.29.2/pyftdi/tests/i2c.py
pyftdi-0.29.2/pyftdi/tests/jtag.py
pyftdi-0.29.2/pyftdi/tests/spi.py
pyftdi-0.29.2/pyftdi/tests/uart.py

Is your intent to license this under MIT or BSD 3-Clause?

eblot, 2019-08-01:
You are right, it needs clarification but I'm not a lawyer and even the LGPL v2 is nearly impossible to define with Python code - at best, nobody agrees.

Let's say it is closer to BSD 3-clause than to MIT, but does it really matters? Do you have a licensing issue with PyFtdi?

eclipseo, 2019-08-01:
It matters because it is ambiguous and for packaging in a distro, we need to clear this ambiguosity, So it would help if you s/MIT/BSD-3-Clause/

eblot, 2019-08-02:
Should be done in 9000c411d.

fabaff, 2019-08-17:
Can you please create a new release? And also push it to PyPI? Thanks.

eblot, 2019-08-19:
Done as 9000c411dfc3db1e1b81f3fa9ea45ec42393dd06 and https://pypi.org/project/pyftdi/0.29.6/

closed 2019-08-19 by eblot