eblot/pyftdi #141: problem in installation of pyftdi

saraazizi76, 2019-08-25:


saraazizi76, 2019-08-25:
Hi i have some problems about installation of pyftdi.
i run these commands:

$sudo apt update
$ sudo apt install python3.6
$ sudo apt install python3-pip
$ sudo apt install python-pip
$ sudo apt install git
$ sudo pip3 install --upgrade setuptools
$ cd ~
$ sudo git clone https://github.com/eblot/pyftdi
$ cd pyftdi
$ sudo -H python3 setup.py install

but i get this error:
Traceback (most recent call last):
File "setup.py", line 34, in 
from setuptools import find_packages, setup
ModuleNotFoundError: No module named 'setuptools'

and guid in http://eblot.github.io/pyftdi/installation.html doesn't work

eblot, 2019-08-25:
> $ sudo apt install python-pip

This one is for Python 2.x series, isn't it?

> $ sudo apt install git
> $ sudo pip3 install --upgrade setuptools

I'm not sure it is a good idea to mix OS managed packages (from deb/apt) with PIP managed packages, especially setuptools which is an endless source of troubles.
`setuptools` are likely provided as `python3-setuptools` on your distribution.

> $ sudo git clone https://github.com/eblot/pyftdi

sudo here? What for?

> ModuleNotFoundError: No module named 'setuptools'
> Do you know what is the problem?

Maybe a mismatch in installation patch.
Do you really need the development version? If not, I would recommend to try installing from PyPi first:

    sudo pip3 uninstall setuptools
    sudo apt-get install python3-setuptools
    sudo pip3 install pyftdi

or alternatively, use a virtualenv

    python3 -m venv ~/pyftdi
    . ~/pyftdi/bin/activate
    pip3 install [-U] setuptools
    pip3 install pyftdi

saraazizi76, 2019-08-27:
> This one is for Python 2.x series, isn't it?
>
>
> I'm not sure it is a good idea to mix OS managed packages (from deb/apt) with PIP managed packages, especially setuptools which is an endless source of [...]

it was solved. thank you

closed 2019-08-27 by saraazizi76