eblot/pyftdi #143: I2C GPIO Resets

zhenqlin, 2019-09-10:
Here's the code I'm using.
https://github.com/eblot/pyftdi/releases/tag/v0.30.0
when I do i2c write, gpio resets.
I'm using FT2232H.
`
from pyftdi.i2c import *
from pyftdi.ftdi import *
i2cCtrl = I2cController()
i2cCtrl.configure('ftdi://::/2')
i2c_slave = i2cCtrl.get_port(0x41)
i2c_io = i2cCtrl.get_gpio()
i2c_io.set_direction(0x0010, 0x0010)
i2c_io.write(0x0010)
i2c_slave.write([0x12, 0x34])
`
1 - BDBUS4, 
2 - SDA
3 - SCL
[image]

eblot, 2019-09-11:
Weird.

I've added this test:

    class I2cIssue143(TestCase):
        """#143.
        """
    
        def test(self):
            url = environ.get('FTDI_DEVICE', 'ftdi://ftdi:2232h/1')
            address = environ.get('I2C_ADDRESS', '0x50').lower()
            addr = int(address, 16 if address.startswith('0x') else 10)
    
            i2c = I2cController()
            i2c.configure(url)
            slave = i2c.get_port(addr)
            io = i2c.get_gpio()
            io.set_direction(0x0010, 0x0010)
            io.write(0x0010)
            io.write(0x0000)
            io.write(0x0010)
            slave.write([0x12, 0x34])

and run it as:

    commit fbfddb33ada5bd4c8a6cee448d2454a019a087e4 (HEAD, tag: v0.30.0)
    Author: Emmanuel Blot <emmanuel.blot@free.fr>
    Date:   Thu Aug 29 19:27:25 2019 +0200

    eblot@spey:~/Sources/Git/github.com/pyftdi
    $ PYTHONPATH=. FTDI_DEVICE=ftdi://ftdi:232h:FT0FMF6V/1 pyftdi/tests/i2c.py
    .
    ----------------------------------------------------------------------
    Ran 1 test in 0.095s
    
    OK

and I observe the expected result... It is not an FTDI2232H but a FTDI232H though, but I do not see why it would make a difference.

[image]

However, something is unexpected on both captures: SCL is kept low between I2C access. There is a new issue here.

eblot, 2019-11-21:
Closing as requested.

Alenlu, 2020-02-06:
Only to comment this issue.
On all devices that I have, I found that the gpio (low byte) is reseting during reading of ACK.
This issue for me was soled to changing the _tristate variable in i2c.py to act similar to _clk_lo_data_hi
@property
def _tristate(self) -> Tuple[int]:
  if self._tristate_use:
     return (Ftdi.SET_BITS_LOW, 
               (self._gpio_low & 0xfe),
               (self.SCL_BIT | (self._gpio_dir & 0x00ff)))
  else:
     return None

in __init__(self) insted of self._tristate I put self._tristate_use = False
and in configure(.....) at the end where except FtdiFeatureError:
instead of setting self._tristate I set self._tristate_use = True

That solve the problem of reseting GPIO (low byte), gpio pins in byte with i2c.
(ft4232h and ft2232h gpio pins GPIOL0-GPIOL3)
Can you confirm that? I use version 0.42.2

eblot, 2020-02-06:
(last comment edited to fix MD syntax)

Alenlu, 2020-02-06:
Tnx eblot

eblot, 2020-02-07:
I need to re-analyse this issue then.

Alenlu, 2020-02-07:
This was the behavior before I made the changes.
Yellow is GPIOL1 and violet is SDA signal on ft2232h. 
If you look at debug string the msg. was 110000a280000122008022fb87
80 00 01 reset the gpio state.
[image]
After the changes the GPIOL1 was low or if set high was high. 
Now the debug msg. is 110000a28020f922008022fb87.

zhenqlin, 2020-02-07:
sorry for late reply. Yes, Alen fixed this issues. Thanks.

eblot, 2020-02-19:
I'm still unable to reproduce this problem. I think I do not understand it. Despite the various graph, I'm not sure to understand the exact sequence.

Do you mean that when you first set AD4 to "high" on each I2C transfer AD4 goes "low" for some time before returning back to "high". Is that it?

eblot, 2020-02-19:
Ok, reproduced, sorry for the trouble.

I should be fixed with an upcoming version, where I2C implementation has been slightly updated.

eblot, 2020-02-19:
Should be fixed in 4e348963ab0f1b8643315e87ee0082452010fec4
(https://github.com/eblot/pyftdi/tree/issue_143)

Not yet delivered to master, as I'm waiting for a feedback using a clock stretching issue to deliver the new version.

Alenlu, 2020-02-20:
tnx

eblot, 2020-02-20:
Should now be fixed in master branch, v0.44.0

closed 2020-02-20 by eblot