eblot/pyftdi #145: serialext: OSError when using readline (read() should have returned a bytes object, not 'bytearray')

jonathangjertsen, 2019-10-01:
With PyFTDI 0.30.3, the following causes an OSError:

```Python
from pyftdi.serialext import serial_for_url

s = serial_for_url(my_ftdi_url, baud_rate, timeout)

line = s.readline()
```

The error reads as follows:

```
Traceback (most recent call last):
...
    line = s.readline()
OSError: read() should have returned a bytes object, not 'bytearray'
```

I see that this is fixed in master (where the `bytearray` gets converted to `bytes` before returning), but this has not been included in any releases yet.

---

If anyone else is having similar problems, the following workaround should fix this specific problem (and is hopefully forwards compatible)

```Python
from pyftdi.serialext import serial_for_url

def pyftdi_readline_fix(ser):
    original_read = serial.read

    def new_read(*args, **kwargs):
        ret_val = original_read(*args, **kwargs)
        if isinstance(ret_val, bytearray):
            return bytes(ret_val)
        else:
            return ret_val
    ser.read = new_read

ser = serial_for_url(my_ftdi_url, baud_rate, timeout)
pyftdi_readline_fix(ser)

line = ser.readline()
```

eblot, 2019-10-01:
Hi @jonathangjertsen,

As the API differs, it is not part of the 0.30+ series, only in the master branch. As I plan to do some other API changes, I'm waiting a bit before releasing any new version.

Your workaround looks nice, thanks.

eblot, 2019-11-21:
0.40 series implement this updated API.

closed 2019-11-21 by eblot