eblot/pyftdi #146: C232HM-DDHSL-0 FTDI Cable: NACK from Slave

nelsoncsc, 2019-10-05:
Hi @eblot,
thanks for the library. I am trying to interface the [C232HM-DDHSL-0 FTDI Cable](https://www.digikey.co.uk/product-detail/en/ftdi-future-technology-devices-international-ltd/C232HM-DDHSL-0/768-1106-ND/2714139?utm_adgroup=Cable+Assemblies&mkwid=sg7dYTYON&pcrid=267270544305&pkw=&pmt=b&pdv=c&productid=&slid=&gclid=CjwKCAjw_uDsBRAMEiwAaFiHa-RuAKWLWcIxy9E_nq-HutLU0cwkxQydzTPNdN3zmEqRdQwvitpnPRoCuF4QAvD_BwE)  to an I2C slave (address is 0x3c).

 After doing an installation of pyftdi on Ubuntu 18.04 I am getting the following error: **pyftdi.i2c.I2cNackError: NACK from slave**

The code is as follows

**>>>from pyftdi.i2c import I2cController***
***>>>i2c=I2cController()***
***>>>i2c.configure('ftdi://ftdi:232h:FT2P33F1/1')***
***>>>slave = i2c.get_port(0x3c)***
***>>>slave.write_to(0x06, b'\x00')**

I shortcut yellow and green wires to get bidirectional SDA and use orange wire as SCL. 
Could you please let me know any clue?

PS: I am able to write and read from slave using a rapsberry pi.
Thank you

nelsoncsc, 2019-10-07:
Hi, just to let everyone know. Adding a resistor of 4k7 to SDA and another 4k7 to SCL actually solve the problem.

closed 2019-10-07 by nelsoncsc

eblot, 2019-10-08:
Yes, I2C is an open collector bus, and FTDI as any other master does not provide the pull up resistors. Maybe I should add this info in the doc.