eblot/pyftdi #148: Reconnect to FTDI

havregn, 2019-10-10:
I wonder if there is a way to reconnect to an FTDI after unplug and replug it. Right now the only workaround I got is restarting my script.

eblot, 2019-10-15:
I'm not sure to understand the exact issue: do you need to manage an error, to automatically reconnect when FTDI device is plugged back, do you have trouble re-opening an FTDI connection to the device, ...?

Skypodo, 2019-12-16:
Sound to me like he has a program running and wants to un- and replug the FDTI device which is almost what I'm trying to do.
Is there a way to do this? My current test looks like this and doesn't work unfortunately.
It does work if my device is connected before my program starts but it does not work if my program is already running and I connect my device.

```
while self.gpio is None:
    sleep(1)
    try:
        print("create controller")
        self.gpio = GpioController()
        print("configure controller")
        self.gpio.configure(self.url, direction=p_direction)
    except:
        print("not connected")
        del self.gpio
        self.gpio = None
print("success!!!")
```

eblot, 2019-12-16:
Flush the local USB cache:

    class HotplugTestCase(TestCase):
    
        def test_hotplug_discovery(self):
            url = environ.get('FTDI_DEVICE', 'ftdi://ftdi:2232h/1')
            ftdi = Ftdi()
            timeout = now() + 5.0  # sanity check: bail out after 5 seconds
            while now() < timeout:
                try:
                    ftdi.open_from_url(url)
                    break
                except UsbToolsError:
                    UsbTools.flush_cache()
                    sleep(0.05)
                    continue
            self.assertTrue(ftdi.is_connected, 'Unable to connect to FTDI')
            print('Connected to FTDI', url)

Skypodo, 2019-12-16:
I was just about to write that :smile:
Thanks for your help @eblot, maybe this is what @havregn is looking for. In my case this works.

Potemkinkuliss, 2019-12-25:
@Skypodo @eblot: @havregn got closed. Thank you very much for helping me with this. Your example solved is for me. Thank you!

eblot, 2019-12-25:
Thanks for the feedback.

closed 2019-12-25 by eblot