eblot/pyftdi #149: serialext parity error detection on RX

dq-gregory, 2019-10-16:
Hi, is the pyftdi serial extension support a mechanism to detect reception of parity error? I am using pyftdi to perform UART transmission with parity check. For the transmission side, I could set it to transmit the parity bit by setting the 'parity' attribute. I am wondering for the receiving side, how I detect the receive of a parity error.

eblot, 2019-10-16:
FTDI HW uses a 2-byte status for each USB RX request. The parity bit is encoded in these bytes.
You can enable FTDI log to get error when a parity mismatch is detected on FTDI HW.
Unfortunately there is no other support for parity checking in PyFTDI, for two reasons:

- I have not thought it could be useful, knowing how limited error detection comes with parity bit,
- A parity bit applies to a single byte, while USB communication usually means that many bytes are send/received at once with a single USB request, and there is no way to know which byte has been detected with a parity mismatch, leaving the parity detection as useful as "there may have been a comm error somewhere" :-)

I'm not even sure how FTDI sets the parity bit in the 2-byte status when several bytes are retrieved at once. I guess it can only be useful when FTDI data is read one byte after another, *i.e.*  one USB cycle for each byte, implying a massive USB overhead.

I guess you can tweak the PyFtdi code to raise an exception when FTDI HW reports a parity error.  Search for `decode_modem_status()` from `ftdi.py` which is the routine that decodes the status bits. It is used in the `read_data_bytes()` to generate the error log message for example. It  also means you need to read one byte after another if you want to detect which byte is in error...
I would strongly suggest to create parity errors to test how effective FTDI HW is to report such errors...

dq-gregory, 2019-10-17:
Thanks for the reply Eblot. I am actually try to use the parity bit to emulate 9-bit data transmission. The parity check is used to get the 9th bit sent by the device that I communicate with. Will take a look into the ftdi.py.