eblot/pyftdi #150: How to use SPI ?

freelancer117, 2019-10-22:
Hello I'm trying to use SPI communication with Pyftdi but I'm struggling.

When I run spi_example.py I see that error :

Traceback (most recent call last):
  File "C:\Users\aguillem\Documents\Python_project\SPI\spi_example.py", line 27, in <module>
    UQDS = COMspi(com)
  File "C:\Users\aguillem\Documents\Python_project\SPI\CommunicationProtocol.py", line 76, in __init__
    self.spi.configure('ftdi://ftdi:2232h/1')
  File "C:\Users\aguillem\AppData\Local\Continuum\anaconda3\lib\site-packages\pyftdi\spi.py", line 315, in configure
    url, direction=self._spi_dir, initial=self._cs_bits, **kwargs)
  File "C:\Users\aguillem\AppData\Local\Continuum\anaconda3\lib\site-packages\pyftdi\ftdi.py", line 481, in open_mpsse_from_url
    vendor, product, index, serial, interface = self.get_identifiers(url)
  File "C:\Users\aguillem\AppData\Local\Continuum\anaconda3\lib\site-packages\pyftdi\ftdi.py", line 311, in get_identifiers
    cls.SCHEME, cls.VENDOR_IDS, cls.PRODUCT_IDS, cls.DEFAULT_VENDOR)
  File "C:\Users\aguillem\AppData\Local\Continuum\anaconda3\lib\site-packages\pyftdi\usbtools.py", line 352, in parse_url
    devices = devclass.find_all(vps)
  File "C:\Users\aguillem\AppData\Local\Continuum\anaconda3\lib\site-packages\pyftdi\ftdi.py", line 392, in find_all
    return UsbTools.find_all(vps, nocache)
  File "C:\Users\aguillem\AppData\Local\Continuum\anaconda3\lib\site-packages\pyftdi\usbtools.py", line 68, in find_all
    devs.update(UsbTools._find_devices(v, p, nocache))
  File "C:\Users\aguillem\AppData\Local\Continuum\anaconda3\lib\site-packages\pyftdi\usbtools.py", line 241, in _find_devices
    raise ValueError('No backend available')
ValueError: No backend available

the errors comes from the line 76 from ComunicationProtocol.py :
spi.configure('ftdi://ftdi:2232h/1')

what is this address, how can I now it ?

I hope someone can help me

[CommunicationProtocol.txt](https://github.com/eblot/pyftdi/files/3755946/CommunicationProtocol.txt)
[spi_example.txt](https://github.com/eblot/pyftdi/files/3755947/spi_example.txt)

eblot, 2019-10-22:
It means PyUSB is unable to find a valid USB library to communicate (`libusb` or the like).

Did you follow the Window specific installation instructions from the documentation (everything is more complex with Windows )? :-)

freelancer117, 2019-10-22:
Ah okay I did nothing about libusb, and yes I'm on windows. I just download the files and yes it look complicated, I'll try it tomorrow.

freelancer117, 2019-10-23:
after several attemps to install libusb I do not understand how to install it.. and I found the pip install here : 
https://pypi.org/project/libusb/
no need to explain how easy it is with pip, but now how can I know if it's correctly installed ? because I'm facing the same error with my python program :-(

eblot, 2019-10-23:
Have you followed the PyFtdi instructions for Windows?

PyFtdi is based on PyUSB which is (one of) the Python interfaces for native libusb. 

https://pypi.org/project/libusb seems to be an alternative to PyUSB. It seems to be a direct wrapper, but is not supported (again, check out the [doc](http://eblot.github.io/pyftdi/installation.html#installation), therefore it will not work with PyFtdi.

libusb configuration is not straightforward under Windows and will never be, AFAICT. This is why [using Zadig](http://eblot.github.io/pyftdi/installation.html#id1) is warmly recommended.

freelancer117, 2019-10-23:
Ah thanks, very easy with that !
No more error now, I will try to deal with the code now. But do you know were to find other SPI example than here https://eblot.github.io/pyftdi/api/spi.html

eblot, 2019-10-23:
What kind of usage example do you need?

freelancer117, 2019-10-23:
[image]

here is my schematic block, I would like to write a python code for the UM232H wich have an FT232H chip. He will be the master and the slave would be the ADUC 814.

eblot, 2019-10-23:
Let me rephrase: what kind of frames do you want to exchange over SPI, or what is missing from the documentation code snippets to fit your needs?

freelancer117, 2019-10-23:
Well I think that's fine actually, I'll let you know soon, I keep the topic open and come back soon.
Thanks a lot !

freelancer117, 2019-10-28:
Everything is fine I can see the signal with an oscilloscope and it's working !

Thanks for your help

closed 2019-10-28 by freelancer117

eblot, 2019-10-28:
Ok, thanks for the feedback.