eblot/pyftdi #151: I2C Missing required positional arguments

pkissel, 2019-10-24:
Hello,

I'm probably missing something simple but when I try the example code I'm getting missing positional arguments errors.

`from pyftdi.i2c import I2cController
i2c = I2cController
i2c.configure('ftdi:///1')`

Which gives the error:

    i2c.configure('ftdi:///1')
TypeError: configure() missing 1 required positional argument: 'url'

If I change this to

`from pyftdi.i2c import I2cController
i2c = I2cController
i2c.configure(url='ftdi:///1')`

Then I get 
    i2c.configure(url='ftdi:///1')
TypeError: configure() missing 1 required positional argument: 'self'

eblot, 2019-10-24:
It is because you are trying to call configure on the class, not an instance of the class. Try

    i2c = I2cController()  # note the parenthesis

then either

    i2c.configure('ftdi:///1') 

or

    i2c.configure(url='ftdi:///1') 

See the examples in the [documentation](http://eblot.github.io/pyftdi/api/i2c.html#quickstart)

pkissel, 2019-10-24:
Silly mistake.  That works, thank you!

closed 2019-10-24 by pkissel