eblot/pyftdi #152: I2C NAK Issue

pkissel, 2019-10-25:
Hello,

I almost have I2C communication working between the PC (I2C master) and a slave device.  I see from the scope waveform the slave acknowledgement, but the I2C function throws a NAK error.

My script is as follows:

> from logging import ERROR, getLogger
from os import environ
from sys import modules, stderr, stdout
from pyftdi.i2c import I2cController, I2cNackError
url = environ.get('FTDI_DEVICE', 'ftdi:///1')
i2c = I2cController()
getLogger('pyftdi.i2c').setLevel(ERROR)
i2c.set_retry_count(1)
i2c.configure(url,frequency=100E3)
port = i2c.get_port(0x40)
try:
    port.read(0)
    print(' read successful!')
except I2cNackError:
    print(' I2cNackError!')
finally:
    i2c.terminate()

[image]

However my script is reaching the i2cNackError and printing that out instead of a successful read.

As I understand with I2C at the ninth clock is when the slave sends an ACK by pulling the SDA line low which is what is happening here.  So why am I getting a NAK error?

pkissel, 2019-10-25:
Nevermind, per the documentation I didn't have the SDA input and output lines tied together.

closed 2019-10-25 by pkissel

ab190001, 2025-11-26:
Me as well --> "Took me way too long with the same problem. Thanks!"