eblot/pyftdi #154: Allow setting of initial configuration

russdill, 2019-11-08:
It's often important to avoid glitches on certain GPIOs. Although the FTDI USB messaging allows initial direction and pin state to be set when configuration mpsse, pyftdi doesn't expose this. Additionally, when changing GPIO direction to output, there is currently no way to set the desired output value first.

eblot, 2019-11-12:
Hi,

The lower API allows this initial settings, but the upper APIs (SPI/I2C) do not expose it, I will have a look at it. I'm not confident that FTDI HW is glitch-free though.

eblot, 2019-11-12:
I think it cannot be always glitch free because the FTDI HW is reset on open, and it is likely to reverse to UART mode if the EEPROM is not configured for another startup mode for example.

Can you check out the `mpsse_initial_config` branch and give it a try? I do not have the SPI/I2C slaves at hand to perform the non-regression tests for now.

eblot, 2020-02-12:
It has been integrated to the official version (80e3f49895dd29dc62d947a19ade0544969f8463 and f873ff431562aa61ae6230d0a199decf6313c9b9)

closed 2020-02-12 by eblot