eblot/pyftdi #155: Fix JTAG

smuecke, 2019-11-13:
As mentioned in my e-mail, `pyftdi.jtag` seems to be outdated as the other parts of the API have changed. My minimal example that throws an exception:

```
from pyftdi.jtag import JtagEngine
f = JtagEngine(trst=True)
f.configure('ftdi:///1')
f.reset() # raises TypeError('Expect a byte array')
```

Also JTAG could use some documentation and examples.
Best regards! ;)

eblot, 2019-11-13:
Hi,

Can you give a try to `jtag_bytearray` branch?

smuecke, 2019-11-13:
Thanks, now I'm getting past `reset()` ☺ However, I'm still getting an exception when trying to write something using `write_dr()`:

```python
f = JtagEngine(trst=True)
f.configure('ftdi:///1')
f.reset()
f.write_dr(b'\x00\x00\x00\x04')
f.write(b'\x00\x00\x00\x00')
```

```
File "./jtag.py", line 12, in <module>
    f.write_dr(b'\x00\x00\x00\x04')
  File "/usr/lib/python3.7/site-packages/pyftdi/jtag.py", line 487, in write_dr
    self._ctrl.write(data)
  File "/usr/lib/python3.7/site-packages/pyftdi/jtag.py", line 288, in write
    out = BitSequence(out)
  File "/usr/lib/python3.7/site-packages/pyftdi/bits.py", line 84, in __init__
    self._init_from_iterable(value, msb)
  File "/usr/lib/python3.7/site-packages/pyftdi/bits.py", line 196, in _init_from_iterable
    raise BitSequenceError("Invalid binary character in initializer")
pyftdi.bits.BitSequenceError: Invalid binary character in initializer
```

When trying with a plain int as argument (`f.write_dr(0x04000000)`) I get this:
```
File "./jtag.py", line 12, in <module>
    f.write_dr(0x04000000)#(b'\x00\x00\x00\x04')
  File "/usr/lib/python3.7/site-packages/pyftdi/jtag.py", line 487, in write_dr
    self._ctrl.write(data)
  File "/usr/lib/python3.7/site-packages/pyftdi/jtag.py", line 295, in write
    self._write_bytes(out[:pos])
  File "/usr/lib/python3.7/site-packages/pyftdi/jtag.py", line 401, in _write_bytes
    bytes_ = out.tobytes(msby=True)  # don't ask...
  File "/usr/lib/python3.7/site-packages/pyftdi/bits.py", line 157, in tobytes
    return bytes_.tolist()
AttributeError: 'bytearray' object has no attribute 'tolist'
```

Seems to be a remnant from Python2.7? What types of input do `write_dr` and `write` expect?
Best regards, and thanks for your quick response! ;)

eblot, 2019-11-13:
Yes it is. Let me check, I have not read this code for over 4 years :-)
I'll get back to you, and try to fix the unit test as well.

eblot, 2019-11-13:
Please pull the latest commit.

smuecke, 2019-11-13:
Brilliant, thanks! ;)

closed 2019-11-19 by eblot