eblot/pyftdi #156: Questions to `Ftdi.set_line_property` and `Ftdi.set_flowctrl`

friedrichromstedt, 2019-11-16:
The docs of my FTDI chip from the vendor of my device request the following configuration:

*   Baudrate 9600;
*   8 data bits, 1 stop bit, no parity;
*   no handshake;
*   no flow control.

From the docs of the class `Ftdi` on http://eblot.github.io/pyftdi/api/ftdi.html I see that:

*   The baudrate can be set by the method `set_baudrate`;
*   the number of data bits, the stop bit and the parity can be set by method `set_line_property`;
*   the flowcontrol mode can be set by method `set_flowctrl`;
*   regarding the handshake I couldn't find a method.

I have a couple of questions and comments related to this:

- [ ] For method `set_line_property`: The *stopbit* count is documented as a *float*; why not as an integer?
- [ ] Furthermore, the bullet list of the section starting with "Parity accepts one of the following strings:" in the docs of `set_line_property` probably has not been written in the ReST source with a blank line between the "... following strings:" sentence and the list, leading to just a paragraph containing asterisks.
- [ ] Method `set_flowctrl`: The docs of the *flowctrl* argument somehow looks messy: it says something like `one of 'hw', ''`. How do I request "no flow control" as requested by my vendor? From the rest of the method's documentation I see that the options should embrace *hardware*, *software* and *no* flow control.
- [ ] The `.. note::` directive in the body of the documentation of `set_flowctrl` looks like if the space between `..` and `note::` has been forgotten.
- [ ] What about the *handshake*?

eblot, 2019-11-17:
## Opening a serial port with PyFtdi

If you do not have any reason to avoid using `pyserial` module, I would recommend you simply use it. `pyftdi` comes with a dedicated protocol handler for `pyserial`, so the standard `pyserial` API may be used to configure and tweak the UART connection.

 See https://eblot.github.io/pyftdi/api/uart.html for details about how to extend `pyserial` to support `pyftdi`, then https://pythonhosted.org/pyserial/shortintro.html#opening-serial-ports for opening and configuring the serial port.

You should not have to deal with the ftdi API to manage a simple UART connection, e.g. for 9600 8N1, it should be as simple as:

    import pyftdi.serialext
    port = pyftdi.serialext.serial_for_url('ftdi:///1', baudrate=9600)

    # 8N1 are the default settings, you can enforce it if you really want to
    port = pyftdi.serialext.serial_for_url('ftdi:///1', baudrate=9600, bytesize=8, parity='N', stopbits=1)

In other words, replace `open` with `serial_for_url` then use `pyserial` API for any other serial related commands.

You can find a working example within the `pyterm.py` file (from `serialext/tests` directory)

Do not mix direct calls to `pyftdi` and calls to `pyserial` API: if you choose the `pyserial` API, which I recommend, always change the serial line settings through the `pyserial` API. It will call `pyftdi` API with the proper settings, when required. 

## PyFtdi API details

* `set_line_property` uses a float for stopbit but do accept an integer. Some weird serial devices use a bit-and-a-half stop bits, so this rather historical setting deals with it. As long as you use `pyserial` API, you never have to call `set_line_property` directly, use *port*`.stopbits` setter
* `set_flow_control` is usuall not be called directly but through the *port*.`rtscts` setter as well. About `hw`, ``:  `hw` means HW flow control, `` (empty string) means NO flow control. SW flow control is no longer supported, as it has never been tested and would be pretty useless any, see below.

### Flow control

Flow control with an FTDI device (*i.e.* FTDI HW, not specific to PyFtdi implementation) is pretty much useless. 
 - SW flow control would be hard to make useful/robust because of the various HW and SW buffer between the host CPU and the USB FTDI slave
 - HW flow control  is managed by the FTDI HW,  but their implementation is buggy: the RTS/CTS flag may be sampled up to 3 bytes after the signal is de-asserted, *i.e.* after some bytes have already been lost. To make it clear: you cannot rely on HW flow control with an FDTI device...

### Handshake

You need to provide more details as there are *many* ways to handshake comm on old serial devices - and RTS/CST are part of handshaking. It looks like vocabulary between very old DTE/DCE communication ports, using now-deprecated signals such as DTR and DSR. Those can be directly managed by the SW (see `pyserial` setters) but I guess that as you need none, you can simply ignore those.

Keeping this ticket open, since I need to clean up the doc related to flow control. Thanks for this report.

eblot, 2019-11-18:
Doc fixed in 330cd61f

friedrichromstedt, 2019-11-29:
After reading quite a bit about serial ports I am left with just one question regarding using `PyFtdi` for my device: Can I just use `Ftdi.set_baudrate` to request the 9600 Bd/s and leave all other settings at their defaults? From what you've written, I see the following:

*   That using 8 data bits, no parity and 1 stop bit is the default;
*   that *no* flow control is the default too, as you said hardware flow control would be "buggy";
*   that normally no handshake is performed.

Am I right with this assumptions?

All this settings would be as requested by my vendor.

There is one more comment on the docs: The description of `Ftdi.set_baudrate` at http://eblot.github.io/pyftdi/api/ftdi.html#pyftdi.ftdi.Ftdi.set_baudrate is not easily readable in the paragraph from "see `baudrate`" until "clock drift (3%)".

Please do not close this Issue yet; I intend to comment more in-detail to the more deep remarks you made.

eblot, 2019-12-02:
> Am I right with this assumptions?

Yes, although for code clarity it is always better to specify default values to help the reader.

> There is one more comment on the docs: The description of `Ftdi.set_baudrate` at http://eblot.github.io/pyftdi/api/ftdi.html#pyftdi.ftdi.Ftdi.set_baudrate is not easily readable in the paragraph from "see `baudrate`" until "clock drift (3%)".

Fixed in 4c57cd88.

friedrichromstedt, 2019-12-20:
From your changes in 4c57cd882b137b2024c5b35a9470e9d5e7063beb I get that using `pyftdi.serialext.serial_for_url(<URL>, ...)` is more top-level than using:

    from pyftdi.ftdi import Ftdi
    Ftdi().open_from_url(<...>)

which is used in https://eblot.github.io/pyftdi/installation.html (Section *Post-installation sanity check*).

Before, I assumed that the `pyftdi.serialext` module *is used by* `pyftdi.ftdi`. Apparently I was wrong.

I got my device working (at least in principle) in the first attempt using `pyftdi.serialext.serial_for_url`.

closed 2020-02-07 by eblot