eblot/pyftdi #158: Regression/API break? from v0.40+: /CS pin 1 not reserved for SPI

j123b567, 2019-11-22:
We are using your other project `pyspiflash` and after just installing it by `pip`, it does not work for us with cryptic exception
`/CS pin 1 not reserved for SPI`
when calling `SerialFlashManager.get_flash_device("ftdi://ftdi/1", cs=1, freq=100e6)`

After some investigation, we found, that it is caused by pyftdi version v0.40+. With version prior v0.40, it works.

You have very invisible little warning, that you perform breaking API changes, but after reading, what may cause problems, non of them seems to apply to this issue.

It seems to me, that it is a regression in new library and not the API break.

Anyway, please, can you reconsider releasing of v0.40+ versions as completely new library (pyftdi2 or so), or provide some backward compatibility, or at least use some beta channel until it is stable? It is not common in python world to do these huge breaking api changes.

eblot, 2019-11-22:
#### Error

I think this error is due to the default count of reserved /CS pins when you create the *SpiController*: it used to be `4`, but there were no strong rationale behind this magical value, so it is now only `1`, as per user requests. Try to initialize *SpiController* with `SpiController(cs_count=2)` (at least `2` to be able to get a port with `cs=1`).

This change is backward compatible with 0.30 series, as `cs_count` already existed, but its default value was `4`.

#### About the API versioning
0.40 (and the beta status of PyFTDI, see `setup.py`) means that the API is not yet stable. I am very sorry for the inconvenience, but there would be too many PyFtdi versions/projects to maintain. 

I would recommend you specify in your project a stable version (*e.g.* `pyftdi => 0.30, < 0.40`) so that any PIP update does not break it if you do not need a newer version but the bug fixes.

j123b567, 2019-11-22:
Thank you for your answer, so should I create issue for `pyspiflash`?
Because there is no way how to access `SpiController` directly and it is also probably incompatible with pyftdi > 0.40

BTW there is also no strong rationale behind value 1...

eblot, 2019-11-22:
> Thank you for your answer, so should I create issue for `pyspiflash`?
Yes, please, I should have updated it.

> BTW there is also no strong rationale behind value 1...

The rationale is that most PyFtdi users only connects a single SPI slave but want to use (existing PCBs, ...) the first available GPIO, *i.e.* `AD4`.

j123b567, 2019-11-22:
Thank you for the fast fix of pyspiflash.

I don't know FTDI chip internals, but isn't it better/possible to allocate all pins on demand and not to prealocate fixed number of e.g. chipselects? So if I use cs=1, then this pin is blocked for SPI and no other pins need to be blocked?

eblot, 2019-11-22:
It could be possible, but would make the API more complex, and there are little chance the HW change during code execution.

eblot, 2019-11-22:
PyFtdi v0.42.0 and PySpiFlash v0.6.1 should be ok.

eblot, 2019-11-25:
Please let me know when you have tested it, so I can close this ticket, thanks.

j123b567, 2019-12-19:
I'm very sorry for late response. We solved this by switching to older libraries and after you fixed this, I don't have access to the hardware to test this. So feel free to close this issue. I can reopen/open issue if something goes wrong, but it can take very long time for me.

eblot, 2019-12-23:
Ok thanks

closed 2019-12-23 by eblot