eblot/pyftdi #162: How to write/exchange bit data?

guitarbbg, 2020-01-14:
I used FT2232 to control a IC vai SPI interface on windows. The IC SPI waveform are 12 bits address and 48 bits data that totally need to write 60 bits in once transaction. 
I can not find how to change transaction unit from byte to bit.
Could you kindly help on it?

BR.
Hank

eblot, 2020-01-15:
I'm afraid there is no API in PyFtdi for this use case for now.
Some SPI slaves accepts more bits than required: have you try to "prepend" 4 pad bits before the address, so that only the last 60 bits are actually kept in the slave shift registers?

guitarbbg, 2020-01-16:
Hi Eblot,

Thanks reply.
I will try your suggestion and update later.
Does any API can early pull high cs pins? maybe that is a working way and I can try it also.

Hank

eblot, 2020-01-16:
There is no API to drive /CS in sync with the SPI bit transmission. You can use the pins not used by SPI as regular GPIOs, but you cannot take control over /CS line.

guitarbbg, 2020-01-20:
Hi Eblot,

Follow your suggestion that SDI "prepend" 4 pad bits before the address and it's working.
Thanks a lot. (:

BR,
Hank

eblot, 2020-01-20:
Could you give a try to https://github.com/eblot/pyftdi/tree/spi_bit_access

I've added experimental support for non-byte aligned SPI transfer.

You need to use a new argument, with you use case: `droptail=4` (see unit tests and SPI API doc). The last 4 bits of the last buffer byte will not be clocked into the SPI slave, so you need to format your data buffer to left-align the data within the buffer.

guitarbbg, 2020-01-31:
Hi Eblot,

Sorry for reply late. I had a trip in holiday. (:
I am a junior python user. Could you kindly let me know how to use the "/spi_bit_access" branch?
I want to verify the branch, but it seems didn't work.

[test process]
1. download ZIP from https://github.com/eblot/pyftdi/tree/spi_bit_access
2. python setup.py build
3. python setup.py install
4. add parameter in my code.
read_buf = spi_handle.exchange(write_buf.to_bytes(8, byteorder='big'), **droptrail=4**).tobytes()
5. execute and return error " exchange() got an unexpected keyword argument 'droptrail' "

Thanks
Hank

eblot, 2020-01-31:
You do not need to install PyFtdi to test a specific version. I would recommend you do not install a development version as the main version as it may contain regressions.

The easiest way is to use git:

     mkdir tmp
     cd tmp
     git clone https://github.com/eblot/pyftdi.git
     cd pyftdi
     git checkout -t spi_bit_access
     PYTHONPATH=. pyftdi/tests/spi.py

should run the unit test for SPI unaligned access (`SpiUnalignedTestCase`)
You do want to edit this test file first to see how to use the new `droptail` argument, it is quite simple. I'd say that your needs matches what is done in `test_bytebit_duplex()`.

I'm not sure what are the `.to_bytes()` calls in your example.

The error you got leads to think you have not retrieved the right Git branch, but it will be easier to diagnose using git than going through a ZIP file that loses all source information.

guitarbbg, 2020-02-04:
[image]

Hi Eblot,
It's working. 
Thank you very much.
The attached is spi waveform for you reference.

Hank

eblot, 2020-02-04:
Ok, thanks for the feedback. I'll merge it to the master branch.

eblot, 2020-02-11:
Delivered as 5a9f1c8a4fd85a085ed949719af5a6fe2fab332c (v0.43)

closed 2020-02-11 by eblot