eblot/pyftdi #163: I2C Clockstretching mode

ElmarJongerius, 2020-01-27:
First of all, great library and thanks for sharing!

I use a FT2232H to communicate over I2C to a custom microcontroller. As some of the requests are a bit intensive for this controller, it will stretch clock cycles. However it seems that the FT2232H is not listening to these stretches, and just continues.

To test this, I have placed a resistor (150r) between ADBUS0 (clk) and the clock line, and monitorred it with my scope on both sides. You can clearly see that the FT2232 is pulling up already while the microcontroller is trying to stretch.

I have also tested with ADBUS5 instead of ADBUS7 but that won`t work.

From earlier issues it looked like clock stretching could be working, but it is not tested a lot. I tried PyFTDI 0.40.6 and 0.42.2, running Python 3.7 on Win7.

I do have a workaround; that is to turn down the clockrate a lot (3000Hz) which allows my MCU to finish without the need for stretching. It's not ideal but it will allow me to continue.

Any help would be welcome!

eblot, 2020-01-28:
Hi,

I should say that I do not have an I2C slave that uses clock stretching, so it is a bit hard to trouble shoot for now.

Simple sanity check as you did not mention it: you enabled the `clockstretching` option as in https://eblot.github.io/pyftdi/api/i2c.html#pyftdi.i2c.I2cController.configure, didn't you?

Which kind of FT2232 are you using? FT2232C, FT2232D or FT2232H?

ElmarJongerius, 2020-01-28:
I'm using a FT2232H (And I'm two ports for I2C).

This is my configuration for the lib.
i2c_ext.configure((self.url + "/1"), clockstretching=True, frequency=1000.0)
(Frequency is set really low now, to make sure my slave can catch up).

The I2C slave I'm using is an LPC824 microcontroller. This is what is shown on my scope (with 150R between the two lines):
[image]
(If I can help debugging or something, please let me know)

eblot, 2020-01-29:
I do not get "FT2232 is pulling up". FTDI does not pull up in I2C mode, pull up is only achieved with the I2C SCK resistor. FTDI only pulls down.

As stated in various FTDI official docs, clock stretching is not officially supported by FTDI HW. It is a hack to re-use the adaptative clock feature designed for ARM JTAG that is expected to allow this feature to work.

Anyway and AFAIU, with clock stretching the FTDI is not supposed to emit a new clock cycle before the slave release the SCL line, but it does not prevent the master (i.e. the FTDI) to release the SCL line.

I'm not sure how to interpret your scope as it depends on the SCL pull up load, but AD0/SCL does not seem to be pulled down by FTDI on the last clock cycle... I would also interpret the last clock cycle on AD7 as follows, as there are 4 different levels on your capture:

[image]

Moreover the time the SCL line stays up (i.e. both master and slave Hi-Z) is constant for all cloick cycles, which seems to indicate that the FTDI had detected and complied with the clock stretching condition of the slave, as it had ignored it, the last SCL up time should be shorter (FTDI SCL up time - slave SCL down time).

ElmarJongerius, 2020-02-18:
It took some time, but I think that I do not agree with you analysis :)
Especially at the point where you did find "FTDI HiZ+Slave down". To be able to measure the actual current I have soldered a 150r resistor between the SCK_in(ADBUS7) (and rest of circuit) and the CLK_out(ADBUS0) pin of the FTDI.

The pullup (10k) is on the side of ADBUS7; and as you can see the level of ADBUS0 is approx 1.5V higher. That mean that the ADBUS0 pin is actually sourcing ~1,5/150 = ~10mA into the bus; while the I2C slave is pulling down.

[image]

eblot, 2020-02-18:
I'm not sure to interpret your issue then.

What is the exact problem? It seems there are several, but I got lost with what you are trying to demonstrate.

ElmarJongerius, 2020-02-18:
The issues I find are:
1. FT2232H pulls the SCL line high (sourcing current), instead of HiZ.
2. FT2232H does not wait for the SCL to be released by slave before doing so. (but it could be caused by issue above)

I am quite sure that this is how the scope image should be interpreted:
Purple line = ADBUS0 (which is the driving side)
Yellow line = ADBUS7 (sensing side)
[image]

Thanks for looking into this! If I should explain more I'm glad to do that.

eblot, 2020-02-18:
I ran a quick check of all the devices I have at hand (about 15 of them). none is using clock stretching, so it's gonna difficult to debug.

What kind of I2C slave device(s) use this mode, if I can find one easy to order and somewhat useful...?

eblot, 2020-02-18:
Ok, so I create a small test breadboard with an FT232H (not an FTDI2232H, this is important, see below), using the following wiring:

[image]

D1 ensures that FTDI can only pull down, and never source Vdd to SCL.
 * when FT232H is used, `AD0` *high* level is measured @ ~ 2.5v (for Vdd = 3.3v, that matches Vdd - D1 voltage drop). This means it does not source output current as expected; **however**
 * when FT232H is forced to be used as a FT2232H, which does not have an open collector output option, `AD0` *high* is measured @ Vdd = 3.3v! -> This is the very issue you have

Whenever R4 is disconnected from `AD0`, i.e. `AD7` (clock sense) is no longer connected to `AD0` but pulled down to GND, PyFTDI reports an error ("*No answer from FTDI*"), as the FTDI HW deadlocks on waiting for SCL being *high* which never happen, so it seems that clock stretching mode - when the digital levels do not get weird values...

I need to check why SCL actually source output when it is not supposed to do so in "FT2232H" mode...

ElmarJongerius, 2020-02-18:
Hey you're great putting in soo much effort!
I use an LPC824 as an I2C slave device which is giving me the problems, but there are more devices which use this. 
I did never use any of these sensors; but they seem to need clock stretching; and they are quite interesting:
https://ams.com/ccs811 (breakout board available at Farnell for instance: https://nl.farnell.com/410386 )
or a BNO022: (https://nl.farnell.com/2491282).

I would gladly send or refund you such a device.

eblot, 2020-02-19:
Ok, after some experimentation, I see no way to fix the issue with a pure software solution:

When MPSSE commands are used, there is no way to create drive the FTDI output in open collector mode; while the trick to emulate the open collector mode is to invert the FTDI pin direction, it is limited:
  - low logical level: FTDI pin as ouput, write 0
  - high logical level: FTDI pin as input
This can be done for prolog and epilog, *i.e.* start and stop sequences, but it cannot be done with a bit/byte transfer, as a pin can either be set as input or output for the full bit/byte sequence, but not changed until the sequence is complete.

This means that there is no way to force SCL to low level for a half clock period, then move it to high-z for the other half clock period where the SCL pull-up resistor should be used.

I think there are some HW solutions, though:
 * Use a FT232H rather than a FT[24]232H device: the former can explicitly be configured to use an open-collector mode, so an output high level on SCL (and SDA) is high-z, not sourced with Vdd
 * Add a diode (with a low drop out^) to `AD0` so that `AD7`/slave SCL cannot be sourced with `AD0`, *e.g.* `D1` on my schematics, `R1` and `R4` are useless for a real board.

  ^: It seems to work with D1 = 1N4148 on my test board, but 0.65v is still high for a "low" level, and some slaves may have issues here.

eblot, 2020-02-20:
Please let me know how it goes with a diode (I've updated the doc with a schema)  and v0.44.0 version.

ElmarJongerius, 2020-02-20:
Ok, I had a quick test with the diode, but that did not work. Even the bitrates did not function properly, so I quickly soldered back my resistor. I will be able to test tomorrow on another board and also take some measurements; unfortunately I didnt have the time yet. I am quite sure that my circuit was OK; and I did use some schottky diode with a Vf of ~0.2V. More news tomorrow!

eblot, 2020-02-20:
Ok...
I think I need to invest into a clock stretching slave then.
BNO055 is available from Amzon @ €32 (similar to Farnell + shipment fee), while it costs €8 on AliExpress but with about 1 month delay - if the covid19 troubles do not even slow this down further...

eblot, 2020-02-20:
Alternatively, using a small µC to emulate a I2C slave might be a better and more flexible solution.

ElmarJongerius, 2020-02-20:
Let me know what I can do to help. I'm sure there are I2C slave example projects available for ARM MCU`s. Eg LPC824 it would be easy to create a small project.

If you would have an Arduino board laying around it should even be easier (although I'm not sure it stretches, I certainly expect them to do so!):
Just add a small delay in the "receiveEvent()" function of the following example:
`// Include the required Wire library for I2C<br>#include <Wire.h>
int LED = 13;
int x = 0;
void setup() {
  // Define the LED pin as Output
  pinMode (LED, OUTPUT);
  // Start the I2C Bus as Slave on address 9
  Wire.begin(9); 
  // Attach a function to trigger when something is received.
  Wire.onReceive(receiveEvent);
}
void receiveEvent(int bytes) {
  x = Wire.read();    // read one character from the I2C
}
void loop() {
  //If value received is 0 blink LED for 200 ms
  if (x == '0') {
    digitalWrite(LED, HIGH);
    delay(200);
    digitalWrite(LED, LOW);
    delay(200);
  }
  //If value received is 3 blink LED for 400 ms
  if (x == '3') {
    digitalWrite(LED, HIGH);
    delay(400);
    digitalWrite(LED, LOW);
    delay(400);
  }
}`

eblot, 2020-02-20:
I think I have a arduino board ... somewhere. I'll try to locate it.

However, how do you enable clock stretching in the above example?
I know Arduino is not very fast, but I2C is far slower... :)

eblot, 2020-02-21:
Ok, I've been busy today with 3D printing flex filament, but I eventually found a bad chinese Arduino Nano clone.

#### Python code

    class I2cClockStrechingGpioCheck(TestCase):
    
        def test_arduino(self):
            self._i2c = I2cController()
            url = environ.get('FTDI_DEVICE', 'ftdi://ftdi:2232h/1')
            self._i2c.configure(url, clockstretching=True)
            # expect Arduino slave @ 0x33
            port = self._i2c.get_port(0x33)
            # send 'Hello'
            port.write(b'Hello')
            # expect "World"
            data = port.read(5)
            print(hexlify(data).decode())

#### Arduino code

    #include "Wire.h"

    void setup() {
       // set up I2C slave @ 0x33
       Wire.begin(0x33);
       // I2C RX
       Wire.onReceive(receiveEvent);
       //I2C TX
       Wire.onRequest(requestEvent);
    }

    // main routine: wait forever
    void loop() {
       delay(100);
    }

    // read and discard each byte
    void receiveEvent(int bytes) {
       (void)Wire.read();
    }

    // write the response
    void requestEvent(void) {
       // force a delay to trigger a clock stretch
       delay(10);
       Wire.write('W');
       Wire.write('o');
       Wire.write('r');
       Wire.write('l');
       Wire.write('d');
    }

Here is the captured chronogram:

[image]

We can clearly see that on the first master read request (arduino write), the master (FTDI) releases SCL ~ 400 µs after the start bit. SCL "out" rises to ~2.5v, which matches the 0.6v of my 1N4148 (I have no Schottky here)

However, the SCL sense signal is pull down by the slave for about ~ 900µs more. The slave then releases SCL. We can see that SCL "out" rises again from 2.5v to 3.3v, which means neither master or slave pulls it down.

Right after the SCL has been released by the slave, we can observe a sequence of 5 * 8 clock cycle, which matches the Python code. We can also see that SDA (from slave) emits `W`,`o`,`r`,`l`,`d`. 

#### Python output

    $ PYTHONPATH=. FTDI_DEVICE=ftdi://:232h/1 pyftdi/tests/i2c.py
    576f726c64 World
    .
    ----------------------------------------------------------------------

So here, data is neither lost nor corrupted, the FTDI master does manage the clock stretching as expected, at least it does so on the very first bit.

Looking at the Arduino's limited `Wire` API, I see no way to generate a clock stretching sequence at arbitrary location in the I2C stream, as the Wire uses internal buffers when the application is not ready to receive them, or generate a 0xFF byte if no data is available on master's request. 

I do not know Arduino that much (I dislike the whole thing...), but if you know some hints to generate other clock stretch events, I can now test them.

eblot, 2020-03-09:
Hi, any news?

ElmarJongerius, 2020-03-10:
Hi, I missed this somehow. Thanks for the message! I have tried different ways earlier to get an Arduino device to stretch, but I found that it was really unreliable because I could get the interrupts at the moment which it should parse.

In your measurements, do you still have the resistor in place, and is this tested on an actual FT2232H? Or is it a 232 (so it seems from the Python part?).

I tried to document my exact issues but got lost in the process. At a certain moment I got SCL pulled low at random times (due to Arduino), but couldnt clearly reproduce what I saw with my LPC and level shifters in between. I will try further if I can at least find some time. Maybe quickest is to try with a different FT232 device.

ElmarJongerius, 2020-03-12:
Could it be a problem that I'm working on both MSSP devices of the FT2232H? I'm using two different I2C busses.
Today I have connected a MCP2221A device to my target circuit instead of the FT2232H and it works like a charm.

eblot, 2020-03-12:
I  really can't tell. I've never used clock stretching...
When I get some spare time, I will check how it behaves with an FT2232H.