eblot/pyftdi #164: MPSSE mode to UART

Stimprod, 2020-01-28:
Hey, thx for this good python module :) 
I've one question: is it possible to switch dynamically between MPSSE mode and UART ? 

Thx in advance.

eblot, 2020-01-28:
I'm not sure to fully understand the question.

You want to close the MPSSE port and re-open it as UART. However the port configuration is lost, and you may experience glitches on FTDI output pins.

Stimprod, 2020-01-28:
Thx for your support.
I have a non conventional application: 
 - I have to be in input PU on RX and TX pins in default (I configure it via eeprom with ft_prog so it's OK)
 - Switch to UART mode to send some data over serial 
 - Return in MPSSE input PU on RX and TX pins

eblot, 2020-01-29:
Sorry, I do not get what "PU" means here.

I'm not sure how FTDI behaves with an alternative EEPROM configuration. FTDI doc is quite sparse on this matter.