eblot/pyftdi #166: Suggestion: add EEPROM write command to enable CBUS Bit-Bang Mode and CBUS pin control API

dmartauz, 2020-01-29:
"Similar to the FT232R, the FT-X series devices (except the FT240X) also support CBUS bit-bang
mode. To use this mode the CBUS pins must first be configured for ‘GPIO’ in the device’s EEPROM,
for example by using FT_Prog. The D2XX commands FT_SetBitMode and FT_GetBitMode are then
used to ‘send’ and ‘receive’ data respectively. "

Source https://www.ftdichip.com/Support/Documents/AppNotes/AN_373%20Bit-Bang%20Modes%20for%20the%20FT-X%20Series.pdf

eblot, 2020-01-29:
It would be a nice addition. I need to buy a FT230X then...

dmartauz, 2020-01-29:
The feature is available on FT232R and other chips as well. However I am not sure if EEPROM content modification is the same for all chips or not.

eblot, 2020-01-31:
On FT232R and FT-X series...

eblot, 2020-02-12:
FT231X ordered...

richey-v, 2020-02-12:
FYI, EEPROM layout and features supported are adapter specific.  FTDI handles EEPROM config with the GUI tool [FT_PROG](https://www.ftdichip.com/Support/Utilities.htm#FT_PROG).  You may be able to extract the information necessary to support various adapters from [D2XX libraries](https://www.ftdichip.com/Drivers/D2XX.htm) for Linux (libftd2xx).

eblot, 2020-02-26:
FT231X received, one less excuse to have a look at this enhancement request. I need to deliver some other changes first.

eblot, 2020-03-12:
Some progress: c537db236eb661b2d5e083e71b05ff7bcca67d50

I've been working on a virtual USB/FTDI nano framework so some CI tasks can be thrown w/ GitHub actions on each commit.

I've added initial support to configure CBUS on FT23*R/H/X.

* List properties, using a virtual FT230x:

        PYTHONPATH=. pyftdi/bin/ftconf.py -V pyftdi/tests/resources/ft230x.yaml\
        -c ? ftdi:///1
          group_1_slew, remote_wakeup, invert_RI, group_1_drive,
          has_usb_version, out_isochronous, chip, group_0_drive, invert_DTR,
          channel_a_driver, self_powered, has_serial, type, invert_DCD,
          cbus_func_0, product_id, vendor_id, group_0_schmitt, invert_TXD,
          in_isochronous, cbus_func_2, group_1_schmitt, suspend_pull_down,
          usb_version, cbus_func_1, invert_RXD, group_0_slew, cbus_func_3,
          invert_RTS, power_max, invert_CTS, invert_DSR

* Show current config:

      PYTHONPATH=. pyftdi/bin/ftconf.py -V pyftdi/tests/resources/ft230x.yaml \
       -v ftdi:///1
        vendor_id: 0x0403
        product_id: 0x6015
        type: 0x1000
        self_powered: False
        remote_wakeup: False
        power_max: 90
        has_usb_version: False
        has_serial: True
        suspend_pull_down: False
        out_isochronous: False
        in_isochronous: False
        usb_version: 0x0000
        manufacturer: FTDI
        product: LC231X
        serial: FT3KMGTL
        channel_a_driver: VCP
        invert_TXD: False
        invert_RXD: False
        invert_RTS: False
        invert_CTS: False
        invert_DTR: False
        invert_DSR: False
        invert_DCD: False
        invert_RI: False
        group_0_drive: False
        group_0_schmitt: 0
        group_0_slew: 0
        group_1_drive: False
        group_1_schmitt: 0
        group_1_slew: 0
        cbus_func_0: TRISTATE
        cbus_func_1: RXLED
        cbus_func_2: TXLED
        cbus_func_3: TRISTATE
        chip: 0x00

* Show supported modes for CBUS0

      PYTHONPATH=. pyftdi/bin/ftconf.py -V pyftdi/tests/resources/ft230x.yaml \
       -c cbus_func_0:? ftdi:///1
         AWAKE, BAT_DETECT, BAT_DETECT_NEG, BB_RD, BB_WR, CLK12, CLK24, CLK6,
         DRIVE0, DRIVE1, I2C_RXF, I2C_TXE, IOMODE, PWREN, RXLED, SLEEP,
         TIME_STAMP, TRISTATE, TXDEN, TXLED, TXRXLED, VBUS_SENSE

* Configure all CBUS for GPIO mode and create a block of data for a new virtual device

       PYTHONPATH=. pyftdi/bin/ftconf.py -V pyftdi/tests/resources/ft230x.yaml \
       -c cbus_func_0:IOMODE -c cbus_func_1:IOMODE \
       -c cbus_func_2:IOMODE -c cbus_func_3:IOMODE \
       -X 9 -s ABCDE ftdi:///1 | head -16
                80 00 03 04 15 60 00 10 80 2d 08 00 00 00 a0 0a
                aa 0e b8 0c 00 00 00 00 00 00 08 08 08 08 00 00
                00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                29 36 d6 c9 01 00 77 f2 b8 68 40 00 00 00 00 00
                00 00 00 00 44 4d 51 38 4c 42 4a 45 00 00 00 00
                0a 03 46 00 54 00 44 00 49 00 0e 03 4c 00 43 00
                32 00 33 00 31 00 58 00 0c 03 41 00 42 00 43 00
                44 00 45 00 00 00 00 00 00 00 00 00 00 00 00 00
                00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                00 00 00 00 00 00 00 00 00 00 00 00 00 00 a0 f7       

Now if this block of EEPROM data is used in a new virtual device:

       $  PYTHONPATH=. pyftdi/bin/ftconf.py -V pyftdi/tests/resources/ft230x_io.yaml -v ftdi:///1
          vendor_id: 0x0403
          product_id: 0x6015
          type: 0x1000
          self_powered: False
          remote_wakeup: False
          power_max: 90
          has_usb_version: False
          has_serial: True
          suspend_pull_down: False
          out_isochronous: False
          in_isochronous: False
          usb_version: 0x0000
          manufacturer: FTDI
          product: LC231X
          serial: ABCDE
          channel_a_driver: VCP
          invert_TXD: False
          invert_RXD: False
          invert_RTS: False
          invert_CTS: False
          invert_DTR: False
          invert_DSR: False
          invert_DCD: False
          invert_RI: False
          group_0_drive: False
          group_0_schmitt: 0
          group_0_slew: 0
          group_1_drive: False
          group_1_schmitt: 0
          group_1_slew: 0
          cbus_func_0: IOMODE
          cbus_func_1: IOMODE
          cbus_func_2: IOMODE
          cbus_func_3: IOMODE
          chip: 0x00

I have NOT tested this generated configuration with a real device, and there is no API to drive CBUS for now.

eblot, 2020-03-18:
This feature has been implemented in the upcoming v0.47, see 551770ef52e9f54fa7cecfd7a5f18390a4ac249c

dmartauz, 2020-05-03:
Thanks for implementing support of CBUS. I will test during next week.

A question - what is the state of CBUS pin after power-up if it is configured as GPIO? Is it pulled high as high impedance input with pull-up resistor?

eblot, 2020-05-06:
CBUS is enabled by EEPROM configuration, so after power up, the state of the CBUS pin is HW-dependent. Please have a look at [AN_184](https://www.ftdichip.com/Support/Documents/AppNotes/AN_184%20FTDI%20Device%20Input%20Output%20Pin%20States.pdf) *FTDI Device Input Output Pin States* if it can help (sorry I lack time by now to check it myself). Let me know if it helps.

eblot, 2020-05-06:
(reopening to track the power up state info)