eblot/pyftdi #167: Can pyftdi be used to read 128  MByte SPI NAND flash?

AsmitaJha08, 2020-01-30:
Hello ,
Currently I am working with 128 MByte SPI NAND Flash :
http://pro0fc108.hkpic1.websiteonline.cn/upload/PN26Q01AWSIUG.pdf.
Wanted to confirm whether pyftdi would work for it. 

Thanks & Regards
Asmita

eblot, 2020-02-05:
I do not get it:
1. You first ask the very same question as [pyspiflash#16](https://github.com/eblot/pyspiflash/issues/16), and I posted a reply on how to proceed.
2. You replied you'd rather go with a FTDI native library which is unrelated to flash management. No problem with that, but from this point, you are using C/C++, not Python, so neither pyspiflash nor pyftdi are relevant.

What are you trying to achieve?

If you want to drive a SPI dataflash using Python, [pyspiflash](https://github.com/eblot/pyspiflash) is an easier start point than pyftdi, as pyspiflash **is dedicated** to drive SPI dataflash, and **does rely** on pyftdi to access slave SPI dataflash using Python language.

To sum up: if you want to use a pure Python stack, pyspiflash should be a good starting point. If you want to use FTDI native library, then you need to look elsewhere, it is likely than other open source libraries have been written to access a SPI flash device using their DLLs.

AsmitaJha08, 2020-02-05:
Hello @eblot ,
Thanks for the response. During my search in the start I had came across pyftdi and libmpsse based libraries, I went through implementation of both and found that starting with python based would be more convenient for me. Hence finally chosen to go with pyspiflash based on pyftdi. Now looking into details so that that I can properly implement properly for NAND Flash because I need to take into consideration the ECC and error block management while accessing NAND flash. Dealing with NAND Flash is more complex than that with NOR Flash.

Thanks and Regards
Asmita

eblot, 2020-02-07:
As EDC/EDC is mostly hidden and internally handled with this NAND flash, the additional work is quite limited.

The direct read sequence of NOR dataflash should be replaced with 1+3 commands:
  0. send the page address to read
  1. poll for ready status and check the error bit
  2. read from cache
  3. send read next page and loop to 1 until complete
This should be quite straightforward.

Write and lock operations are also pretty similar to what's implemented for some of NOR flash devices.

eblot, 2020-02-11:
I think this does not really belong to PyFtdi, closing as PyFtdi provides the required feature to implement this driver.

closed 2020-02-11 by eblot