eblot/pyftdi #169: Missing LICENSE file

FFY00, 2020-02-22:
`BSD` is refeered in the metadata but this is not legally valid. The code is not distributed under any license, which makes it technically proprietary and so it can't be distributed by distributions. I want to package this for ARchlinux, could you add a license file to the repo? Thank you.

eblot, 2020-02-22:
> BSD is refeered in the metadata but this is not legally valid.

The license is documented in `pyftdi/doc/licenses.rst` and I have not duplicated it into a top-level `LICENSE` root file, if that is the point. The code is distributed under BSD 3-clause, as documented in this file.

I guess you can generate a `LICENSE` root file with something like:

    rst2man.py pyftdi/doc/licenses.rst | groff -mandoc -Tascii | sed 's/^  */  /g'

BTW,  from a vocabulary standpoint, I do not think there is such a "legal" thing that tells that a license should be placed into a `LICENSE` file and [nothing else](https://help.github.com/en/github/creating-cloning-and-archiving-repositories/licensing-a-repository#determining-the-location-of-your-license) :-)

FFY00, 2020-02-22:
Agh, sorry, missed it.

closed 2020-02-22 by FFY00

eblot, 2020-02-22:
No problem :-)