eblot/pyftdi #170: Building wheel fails due to wrong license-file in setup.cfg

JKitok, 2020-02-24:
setup.cfg points to the license-file which was renamed in the latest commit:
https://github.com/eblot/pyftdi/commit/ff9e5352e8c3937663c1b4fc162c878be0aff65a

The commit did not change the pointer in setup.cfg.

This is the last few lines of output from our Jenkins-instance that tries to build the wheel (on Windows, Python 3.7 64-bit):

```
 running egg_info
  writing pyftdi.egg-info\PKG-INFO
  writing dependency_links to pyftdi.egg-info\dependency_links.txt
  writing requirements to pyftdi.egg-info\requires.txt
  writing top-level names to pyftdi.egg-info\top_level.txt
  warning: Failed to find the configured license file 'pyftdi/doc/licenses.rst'
  reading manifest file 'pyftdi.egg-info\SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no previously-included files found matching 'README.md'
  writing manifest file 'pyftdi.egg-info\SOURCES.txt'
  Copying pyftdi.egg-info to build\bdist.win-amd64\wheel\.\pyftdi-0.44.1-py3.7.egg-info
  running install_scripts
  creating build\bdist.win-amd64\wheel\pyftdi-0.44.1.data
  creating build\bdist.win-amd64\wheel\pyftdi-0.44.1.data\scripts
  copying build\scripts-3.7\ftconf.py -> build\bdist.win-amd64\wheel\pyftdi-0.44.1.data\scripts
  copying build\scripts-3.7\ftdi_urls.py -> build\bdist.win-amd64\wheel\pyftdi-0.44.1.data\scripts
  copying build\scripts-3.7\i2cscan.py -> build\bdist.win-amd64\wheel\pyftdi-0.44.1.data\scripts
  error: [Errno 2] No such file or directory: 'pyftdi/doc/licenses.rst'
  
  ----------------------------------------
  Failed building wheel for pyftdi
```

eblot, 2020-02-24:
Oops. I really need to set up a CI with this project. Thanks for the report.

eblot, 2020-02-24:
Should be fixed in 3ab9a227 (and another related issue in 9a2cfd90)

closed 2020-02-24 by eblot

JKitok, 2020-02-24:
Confirmed that 0.44.2 works on our end. Thanks for the quick fix!