eblot/pyftdi #172: device can't be opened under windows

CCHLENG, 2020-02-24:
Hi! thank you for pyfdti.
i want to use ft232h by python.
i installed pyfdti under windows, and set the driver follow the document(by zadig).

[image]

it can work like this:
```cmd
C:\Users\HAN>python
Python 3.6.5 (v3.6.5:f59c0932b4, Mar 28 2018, 17:00:18) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from pyftdi.ftdi import Ftdi
>>> Ftdi().open_from_url('ftdi:///?')
Available interfaces:
  ftdi://ftdi:232h:0:1/1   (Single RS232-HS)

Please specify the USB device
```
but when i want to open the device:
```cmd
C:\Users\HAN>python
Python 3.6.5 (v3.6.5:f59c0932b4, Mar 28 2018, 17:00:18) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from pyftdi.ftdi import Ftdi
>>> a = Ftdi().open_from_url('ftdi://ftdi:232h:0:1/1')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\HAN\AppData\Local\Programs\Python\Python36\lib\site-packages\pyftdi-0.44.1-py3.6.egg\pyftdi\ftdi.py", line 434, in open_from_url
    self.open_from_device(device, interface)
  File "C:\Users\HAN\AppData\Local\Programs\Python\Python36\lib\site-packages\pyftdi-0.44.1-py3.6.egg\pyftdi\ftdi.py", line 487, in open_from_device
    self._set_interface(config, interface)
  File "C:\Users\HAN\AppData\Local\Programs\Python\Python36\lib\site-packages\pyftdi-0.44.1-py3.6.egg\pyftdi\ftdi.py", line 1764, in _set_interface
    if self.usb_dev.is_kernel_driver_active(self.index - 1):
  File "C:\Users\HAN\AppData\Local\Programs\Python\Python36\lib\site-packages\pyusb-1.0.2-py3.6.egg\usb\core.py", line 1098, in is_kernel_driver_active
  File "C:\Users\HAN\AppData\Local\Programs\Python\Python36\lib\site-packages\pyusb-1.0.2-py3.6.egg\usb\backend\libusb0.py", line 632, in is_kernel_driver_active
  File "C:\Users\HAN\AppData\Local\Programs\Python\Python36\lib\ctypes\__init__.py", line 361, in __getattr__
    func = self.__getitem__(name)
  File "C:\Users\HAN\AppData\Local\Programs\Python\Python36\lib\ctypes\__init__.py", line 366, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: function 'usb_get_driver_np' not found
>>>
```
and i installed the `libusb-win32-devel-filter-1.2.6.0.exe` ,follow this [question](https://stackoverflow.com/questions/48253119/pyftdi-finds-device-under-os-x-but-not-windows).

but it still show `AttributeError: function 'usb_get_driver_np' not found`..

what can i do? thank you.

CCHLENG, 2020-02-24:
now i tried these:
```cmd
C:\Users\HAN>python
Python 3.6.5 (v3.6.5:f59c0932b4, Mar 28 2018, 17:00:18) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb
>>> import usb.util
>>> dev = usb.core.find(idVendor=0x0403, idProduct=0x6014)
>>> print(dev)
DEVICE ID 0403:6014 on Bus 000 Address 001 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x200 USB 2.0
 bDeviceClass           :    0x0 Specified at interface
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :   0x40 (64 bytes)
 idVendor               : 0x0403
 idProduct              : 0x6014
 bcdDevice              :  0x900 Device 9.0
 iManufacturer          :    0x1 FTDI
 iProduct               :    0x2 Single RS232-HS
 iSerialNumber          :    0x0
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 500 mA ==================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0x20 (32 bytes)
   bNumInterfaces       :    0x1
   bConfigurationValue  :    0x1
   iConfiguration       :    0x0
   bmAttributes         :   0x80 Bus Powered
   bMaxPower            :   0xfa (500 mA)
    INTERFACE 0: Vendor Specific ===========================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :   0xff Vendor Specific
     bInterfaceSubClass :   0xff
     bInterfaceProtocol :   0xff
     iInterface         :    0x2 Single RS232-HS
      ENDPOINT 0x81: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x0
      ENDPOINT 0x2: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x2 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x0
>>>
```

eblot, 2020-02-24:
Have you strictly followed the installation instructions from https://eblot.github.io/pyftdi/installation.html#windows ?

CCHLENG, 2020-02-25:
[image]
yes, i strictly followed instructions.

eblot, 2020-02-25:
Your screen does not match the installation instructions, step 3:

  *  *3. Select **libusb-win32** (not **WinUSB**) in the driver list.*

I'm confused.

CCHLENG, 2020-02-25:
sorry.. the screenshot is wrong, i can make sure that i use `libusb-win32`.
now i try to install pyfdti on another computer, and it works !
but it still can't work on the old one...

another question about GPIO:  
how can i use the GPIO ACBUS0-7 on my ft232H? Use  SPI API `pyftdi.spi.SpiGpioPort`?

eblot, 2020-02-25:
Additional note: it seems you have not installed PyUSB from an official package but a [development version](https://github.com/pyusb/pyusb/issues/298). 

Please reinstall PyUSB from the latest tagged version before resuming with install troubleshooting, as PyFtdi has not been yet tested with this development version.

Quick check: from the installed PyUSB installation directory, open `usb/core.py`. If the first lines are:

     # Copyright (C) 2009-2014 Wander Lairson Costa

you're good to go. However if it starts with

    # Copyright 2009-2017 Wander Lairson Costa
    # Copyright 2009-2020 PyUSB contributors

You've installed a development version.

CCHLENG, 2020-02-25:
ok! it works after i reinstall the pyusb. thank you very much!

eblot, 2020-02-25:
> now i try to install pyfdti on another computer, and it works !
> but it still can't work on the old one...

That's the magical world of M$ Windows, you never know what's actually running.

Try uninstalling everything you've installed (included all the stuff from the slashdot entry), and try again (with the usual Windows mess: reboot, shake the computer, throw it through the window...)

> another question about GPIO:
> how can i use the GPIO ACBUS0-7 on my ft232H? Use SPI API `pyftdi.spi.SpiGpioPort`?

FTDI's BITBAND mode only address lower pins (AD0..AD7), this is a HW limitation.
However, FTDI's MPSSE mode enables access to all pins (AD0..AD7, AC0..AC7) but PyFtdi does not provide yet a direct GPIO access for this mode. You can nevertheless access these pins though the GPIO ports of I2C and SPI mode. I would suggest to use I2C rather than SPI - if you not need SPI feature, as it would free one more pins:
On FT232H:
  * I2C: AD3..AC7 available as GPIOs
  * SPI: AD4..AC7 available as GPIOs

There is no available mode that gives access to all pins at once (AD0 .. AC7) as GPIOs for now, although it is planned.

CCHLENG, 2020-02-25:
ok ! thank you for your patience : )

closed 2020-02-25 by CCHLENG

eblot, 2020-02-25:
> ok ! thank you for your patience : )
With Windows, I have none :P

You're welcome. I also learned/discovered that PyUSB development has resumed after a long stall, and that the current versioning scheme is likely to cause more issues if the near future.

Thanks.

eblot, 2020-02-25:
For the record, I've logged an issue for the initial error you reported: https://github.com/pyusb/pyusb/issues/299