eblot/pyftdi #173: Simultaneously using UART and GPIO on the same port

langestefan, 2020-02-27:
Hi,

My HW setup currently features a 4232H FTDI IC. From the datasheet, it supports MPSSE on ports A and B. This should allow me to use both UART and GPIO simultaneously on a single port.

If I understand correctly, this usecase is not supported by PyFtdi:

[https://eblot.github.io/pyftdi/api/gpio.html](https://eblot.github.io/pyftdi/api/gpio.html)
> This mode is mutually exclusive with advanced serial MPSSE features, such as I2C, SPI, JTAG, …
> 
> If you need to use GPIO pins and MPSSE interface on the same port, you need to use the dedicated API. For now, this shared mode is only supported with the SPI API. 

Are there any plans to support this in the future?

eblot, 2020-02-27:
Hi,

I think it is somewhat a duplicate of #62 and also related to #164.

As far as I can tell, there is no equivalent mode for this feature (UART+ GPIO) than it exists with I2C and SPI: I2C, SPI and JTAG all uses MPSSE protocol which is mutually exclusive with UART mode. UART mode is also mutually exclusive with BITBANG mode.

FT2232H and FT4232H do not support CBUS GPIO mode either. So I might have missed something from the FTDI datasheet, but I do not get how this lead to "*This should allow me to use both UART and GPIO simultaneously on a single port.*"

If you use a FT4232H, there is 4 ports. You can use any of them as UART, and you can use the two first ones as MPSSE. I never tried to use a combination such as BITBANG mode on one port and UART mode on another port, but it should work, at least for some combinations of them.

However I do not think it is possible to use UART + GPIO on the very same port, can you pin point which datasheet / pages documents this specific use case? Thanks.

langestefan, 2020-02-27:
Sorry I am misunderstanding the documentation I think. I was under the impression that MPSSE was also able to emulate UART + GPIO, as those functions are on seperate pins. Thanks for clarifying, I will close the issue.

eblot, 2020-02-27:
I think you can use still MPSSE on interface 1 and BITBANG on interface 2, i.e.

* AD0..AD7: UART
* BD0..BD7: GPIO

Note there are a couple of UART pins that can be used as GPIO while in UART, but you cannot chose which one and their direction, see FT4232H datasheet, section 3.4.1:
  * RTS and DTR can be used as output
  * CTS, DSR, DCD and RI  can be used as input
If you do not need these pins for UART communication.

langestefan, 2020-02-27:
> I think you can use still MPSSE on interface 1 and BITBANG on interface 2, i.e.
>
> * AD0..AD7: UART
> * BD0..BD7: GPIO
>
> Note there are a couple of UART pins that can be used as GPIO while in UART, but [...]

Thanks a lot! However I don't see a section 3.4.1 in FT4232H datasheet. Do you perhaps mean section 3.1.4?
[image]

Would I be able to read/write DSR, DTR independently from my python application? Can you maybe point me to the API function that would allow that?

eblot, 2020-02-27:
Sorry I had the v2.1 datasheet. I kept the old datasheets as FTDI removed some useful sectiond from the earlier versions... For current v2.6 version, you right it is 3.1.4.

I'm not sure what you meant by " independently from my python application". You cannot use two applications at once to drive the very same port. However you may have two applications, each one addressing a dedicated port of the same FTDI device.

Please clarify what you meant, I'm a bit lost here.

langestefan, 2020-02-27:
No problem, I understand what you mean now.

By 'independently' I mean that I want to have async UART TX/RX + control of CTS/RTS on the same port. I found that I can use this API call to open a pyserial port instance:
`pyftdi.serialext.serial_for_url(url, **options)`

Would this enable me to use API calls from pyserial?

My application needs control of 4 simultaneous UART ports + I need to control/read at least 2 signals per port.

eblot, 2020-02-27:
I'm now not sure what you actually meant:
- you want to use UART + RTS/CTS to perform HW flow control on each port OR
- you want to use UART (RX/TX only) and use RTS/CTS pin to perform random GPIO access?

eblot, 2020-02-27:
Yes `serial_for_url` is the way to go (`pyftdi.serialext.serial_for_url` is mostly a small wrapper). PySerial will use PyFtdi to access the UART with this extension. Once the UART port is open with the `ftdi://` syntax, you only use PySerial to access the UART and the extra UART pins.

All control pins are available as serial port [attributes](https://pythonhosted.org/pyserial/pyserial_api.html#serial.Serial.rts), *i.e.* Python getter/setter.
 properties. Behind the scene, `pyserial` will call `pyftdi`.

langestefan, 2020-02-27:
> Yes `serial_for_url` is the way to go (`pyftdi.serialext.serial_for_url` is mostly a small wrapper). PySerial will use PyFtdi to access the UART with this extension. Once the UART port is open with [...]

Awesome! Thanks for the explanation. It would be very helpful for me and perhaps others if you could add this information to: [https://eblot.github.io/pyftdi/api/uart.html](https://eblot.github.io/pyftdi/api/uart.html)

As this was not clear to me from the documentation. 

> you want to use UART (RX/TX only) and use RTS/CTS pin to perform random GPIO access?

Well in fact my preferred choice is to use UART RX TX + GPIO on a single port. But as you have explained that this is not possible, using RTS/CTS to perform random GPIO access would be the second best option.

eblot, 2020-02-27:
Ok for the doc, I'm keeping this ticket opened till its done. Thanks for the suggestion.

eblot, 2020-03-08:
Documented in 50777a7349f04199a48512b7aa81e19a6666c0f8

closed 2020-03-08 by eblot