eblot/pyftdi #174: Alternative for in_waiting

langestefan, 2020-03-04:
Hi,

Thanks for your work on pyftdi, it has been a great help. My application is using serialext. I am using a driver that relies on in_waiting for reading data in a non blocking way. In protocol_ftdi.py I read that it is not implemented in serialext and therefore always returns 0:

    @property
    def in_waiting(self):
        """Return the number of characters currently in the input buffer."""
        # not implemented
        return 0

Is there any alternative to replace the functionality of in_waiting? Are there any plans to implement this in the future?

eblot, 2020-03-04:
There is no way to know how many bytes are available in the HW FIFO of an FTDI, but to retrieve them. This is why this API is not implemented, because there is no way to implement it in a useful way.

If you are using serialext, that is the PySerial API, simply define a timeout (`port.timeout = ...`). If no byte is available after the specified delay, the `read()` call returns with an empty buffer. 

Common delays range from 1 ms to 250 ms.

langestefan, 2020-03-06:
Thanks for your suggestion. I have re-implemented that part of the driver and it is now working. 

One more thing, I am struggling to understand the `set_direction()` method. The description reads:

```
pins (int) – which GPIO pins should be reconfigured

direction (int) – a bitfield of GPIO pins. Each bit represent a GPIO pin, where a high level sets the pin as output and a low level sets the pin as input/high-Z.
```

But what should I pass for `pins`? Is it also a bitfield like `direction` ? Does it override other pin functions, so do I have to keep a cache? Does direction also override values that were set earlier?

An example on how to use this function would be very much appreciated.

eblot, 2020-03-06:
I'm not sure which of PyFtdi's APIs you are referring to.

You should not have to use at all `Ftdi` instance if you are using serial mode, as the API for UART is PySerial, not PyFtdi low-level API. In this case, as I said earlier, the function and direction of each pin is "hard-coded" by the HW, so you cannot change their functions, and you do not have to configure them either.

Please provide details or an example of what you are trying to achieve, it is a bit fuzzy from my side up to now. Thanks.

langestefan, 2020-03-06:
Sorry, I will try to be more clear. What I am typing now is unrelated to this particular issue thread but I don't know where else I can ask. 

I have a FTDI IC with 4 ports. On one port I am using pyftdi's serial extension for UART. I am aware that it uses pyserial's API, which is what I am calling (but only for TX/RX). 

Because I am controlling a device I also need GPIO. I have decided not to use pyserial for that but instead use pyftdi on a second port and dedicate that port to GPIO. So on a second port I am using the GPIO API from pyftdi, for which I need the `set_direction()` method. 

Right now I am struggling to use `set_direction()` because of the reasons in my previous comment.

eblot, 2020-03-06:
Ok, much clearer. So:

GPIOs are accessed as integers, where b0 is the lowest pin and bN is the highest one.
On FT4232H, there are four interfaces, that is four "narrow" 8-bit ports, which means:
*  AD0/BD0/CD0/DD0:  b0:  1<<0 = 0x01
*  AD1/BD1/CD1/DD1:  b1:  1<<1 = 0x02
* ...
* AD7/BD7/CD7/DD7:  b7:  1<<7 == 0x80

Pins are therefore read and written as 8-bit integer. When read/write, 1 means high level (Vdd, typ. 3.3v), 0 means low-level (GND)

Direction argument use the same notation, where 1 means output, 0 means input.

Assuming you are using GPIO on the second interface (`ftdi://:4332h/2`), that means BD0...BD7, as an example:
 * direction 0x76 = 0b 0111 0110 means 0b `I O O O I O O I`:
     BD0, BD3, BD7: input
     BD1-BD2, BD4-BD6: output
 * reading the port, only BD0, BD3, BD7 are meaningful, *i.e.*
     `gpio.read()` can be seen as 0b `I _ _ _ I _ _ I` when X is the actual input signal, `_` should be ignored, so if BD0 is tied to GND and BD3, BD7 are tied to Vdd, you should read
      `gpio.read() & direction = 0x88`
 * writing the port, only BD1-BD2, BD4-BD6 are meaningful, *e.g.*
    if you want to set the high nibble to 1 and the low nibble to 0, that is `0xF0` 
    `gpio.write(0xF0 & direction) = gpio(0x70)`, that is:
     * BD0: ignored, input
     * BD1, BD2: GND
     * BD3: ignored, input
     * BD4, BD5, BD6: Vdd
     * BD7: ignored, input

   You can define the direction and the initial value on initialisation through the `configure()` method or after initialisation with the `set_direction()` (direction) method and the `write()` method (output value).

Note that writing a `0` to an input pin is ignored, but writing `1` to an input pin will trigger an `ValueError()` exception, *i.e.* when writing a new output value, be sure to mask the output value to mute the input pins, for example with `gpio.write(value & gpio.direction)`

eblot, 2020-03-06:
I've tried to document it here: https://eblot.github.io/pyftdi/gpio.html

langestefan, 2020-03-06:
This is amazing! Thank you so much! I managed to get everything up and running now, thanks to your explanation :)

closed 2020-03-06 by langestefan

eblot, 2020-03-06:
👍🏼Thanks!

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>