eblot/pyftdi #177: _find_devices fails to find all devices on windows

FelixdeNeve, 2020-03-19:
Hi,
Thanks for your work on this library. I'm currently working on a project that uses several FT4232H devices (all with blank EEPROMs) connected to the host PC running windows via a USB hub.
When I use the Ftdi.show_devices() only one of the devices is listed (each of the interfaces are given so it returns 4 URLS). In addition, because the program does not recognise that the other urls exist, I can only open the first device with the url (other devices can be opened as pyusb objects). Using a linux machine all of the devices are correctly listed. 
I found that replacing the _find_devices() function with usb.core.find() causes the program to behave as expected.

eblot, 2020-03-19:
Can you provide some info about how you configured your devices under Windows? Even some screenshots of Zadig interfaces. Thanks.

FelixdeNeve, 2020-03-19:
The devices were configured with Zadig such that the composite parents became LibUSB Win32 Devices.
Here's a zadig screenshot for the configured devices:
[image]

eblot, 2020-03-19:
Ok thanks.
I'll try to reproduce the problem (knowing  I deeply hate Windows...)

FelixdeNeve, 2020-03-19:
Thanks very much!

eblot, 2020-03-19:
Ok, I do not have two FT4232H available, but I found 2 FT232H with no EEPROMs.
On macOS, this gives:

    $ pyftdi/bin/ftdi_urls.py
    Available interfaces:
      ftdi://ftdi:2232:14:12/1   (Dual RS232-HS)
      ftdi://ftdi:2232:14:12/2   (Dual RS232-HS)
      ftdi://ftdi:2232:14:13/1   (Dual RS232-HS)
      ftdi://ftdi:2232:14:13/2   (Dual RS232-HS)

What does it show on Windows (while I upgrade the VM, Windows 10, ...)?

FelixdeNeve, 2020-03-19:
This is similar to what I would get on Linux. On Windows I would expect it to show just: 
  ftdi://ftdi:2232:14:12/1   (Dual RS232-HS)
  ftdi://ftdi:2232:14:12/2   (Dual RS232-HS)

eblot, 2020-03-19:
and what do you actually get?

FelixdeNeve, 2020-03-19:
Here's what I get:

`from pyftdi.ftdi import Ftdi`
`Ftdi.show_devices()`
` Available interfaces:`
` ftdi://ftdi:4232:0:6/1   (Quad RS232-HS)`
` ftdi://ftdi:4232:0:6/2   (Quad RS232-HS)`
` ftdi://ftdi:4232:0:6/3   (Quad RS232-HS)`
` ftdi://ftdi:4232:0:6/4   (Quad RS232-HS)`

eblot, 2020-03-19:
Ok, I think I found the culprit. Working to fix this.

eblot, 2020-03-19:
c:\Users\eblot\AppData\Local\Programs\Python\Python37\Scripts>ftdi_urls.py
    Available interfaces:
      ftdi://ftdi:2232:0:1/1   (Dual RS232-HS)
      ftdi://ftdi:2232:0:1/2   (Dual RS232-HS)
      ftdi://ftdi:2232:0:2/1   (Dual RS232-HS)
      ftdi://ftdi:2232:0:2/2   (Dual RS232-HS)

eblot, 2020-03-19:
It should be fixed. Please re-open with it still does not work with FT4232H

commit 55cce784 (2020-03-19): Fix #177 (unable to enumerate all devices w/o serial number on Windows)
closed 2020-03-19 by eblot