eblot/pyftdi #178: Add FTDI Vendor TIAO USB Multi-Protocol Adapter (TUMPA)

grymoire, 2020-03-22:
This FT2232H board has vendor ID 0x8a98
This patch fixes the problem, I believe. Otherwise, the find_urls.py returns nothing.

--------------------------
--- ftdi.py.orig	2020-03-22 13:37:57.919150970 -0400
+++ ftdi.py	2020-03-22 10:56:17.390397876 -0400
@@ -82,6 +82,7 @@
              '2232': 0x6010,
              '2232d': 0x6010,
              '2232h': 0x6010,
+             '2232h': 0x8a98,
              '4232': 0x6011,
              '4232h': 0x6011,
              '230x': 0x6015,
@@ -93,6 +94,7 @@
              'ft2232': 0x6010,
              'ft2232d': 0x6010,
              'ft2232h': 0x6010,
+             'ft2232h': 0x8a98,
              'ft4232': 0x6011,
              'ft4232h': 0x6011,
              'ft230x': 0x6015,
-------------------------------------

Also - this needs to be added to 11-ftdi.rules
SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="8a98", GROUP="plugdev", MODE="0666"

eblot, 2020-03-23:
Hi,

FTDI exposes an API to add custom VID/PID, see https://eblot.github.io/pyftdi/urlscheme.html#supporting-custom-usb-vendor-and-product-ids

There is an endless list of fancy VID/PID for USB devices, and any user can change the VID/PID through the FTDI EEPROM, this is why only FTDI devices are listed in PyFtdi.

I'm keeping this ticket open till I update the doc about `11-ftdi.rules` with custom VID/PID, thanks for reporting this.

eblot, 2020-03-23:
Documentation improved in f312c073ca4367621b7406fc1c6a50f7a5e6c006.

grymoire, 2020-03-23:
Understood. Thank you. I have 5 different FTDI[2]232H boards from different vendors, and the TUMPA was the only one that caused a problem. It would be nice to have a way to fix this without modifying source.

eblot, 2020-03-23:
Which source? Your application's one?
You can either write a small wrapper to subclass `Ftdi` to add your own VID/PID, or change the EEPROM content to use FTDI's PID if the board has nothing specific: keep the manufacturer name to quickly identify the device, but revert the PID to what's TUMPA likely is: a mere FTDI expansion IO board, if you do not need this very custom PID.

grymoire, 2020-03-23:
In this case, I wanted to run ftdi_urls.py and i2cscan.py, and the new documentation is not clear what I have to modify to get this to work with TUMPA's licensed product ID. The TUMPA board is documented [here](https://www.tiaowiki.com/w/TIAO_USB_Multi_Protocol_Adapter_User%27s_Manual)

eblot, 2020-03-23:
Ok, much clearer, maybe it would useful to add some option switches to these tools in order to add specific VID/PID then?

eblot, 2020-03-23:
Something like:

    $ PYTHONPATH=. pyftdi/bin/ftdi_urls.py -P 0x403:0x8a99 -P 0xcafe:0xbabe \
        -V pyftdi/tests/resources/custom_vidpid.yaml
    Available interfaces:
      ftdi://ftdi:0x8a99:TIAO_USB_1/1           (TUMPA)
      ftdi://0xcafe:0xbabe:SomeRandomNumber/1   (Whatever)

eblot, 2020-03-23:
Should be implemented in v0.48.3 (c268dc94f7eb1db826c888df8421496b1edc3d0b)

See https://eblot.github.io/pyftdi/tools.html#option-p

closed 2020-03-23 by eblot

grymoire, 2020-03-23:
Excellent!