eblot/pyftdi #179: I2cNackError with FT232H+TSL2591

sethpoh, 2020-03-31:
Hi all, 

I am trying to use a FT232H breakout board (Adafruit) as an I2C <-> USB convertor to connect to a TSL 2591 light sensor (Adafruit). The D1 and D2 pints are shorted for I2C operation. I am using the TSL2591 and Blinka library in Circuitpython. The blinka library ultimately uses pyftdi. However when I ran the sample test script for the sensor. I got the following error - 

Traceback (most recent call last):
  File "tsl2591_simpletest.py", line 14, in <module>
    sensor = adafruit_tsl2591.TSL2591(i2c)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_tsl2591.py", line 115, in __init__
    if self._read_u8(_TSL2591_REGISTER_DEVICE_ID) != 0x50:
  File "/usr/local/lib/python3.7/dist-packages/adafruit_tsl2591.py", line 128, in _read_u8
    i2c.write_then_readinto(self._BUFFER, self._BUFFER, out_end=1, in_end=1)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_bus_device/i2c_device.py", line 156, in write_then_readinto
    in_end=in_end,
  File "/usr/local/lib/python3.7/dist-packages/busio.py", line 92, in writeto_then_readfrom
    in_start=in_start, in_end=in_end, stop=stop)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/ft232h/i2c.py", line 35, in writeto_then_readfrom
    relax=True)
  File "/usr/local/lib/python3.7/dist-packages/pyftdi/i2c.py", line 181, in exchange
    readlen, relax=relax)
  File "/usr/local/lib/python3.7/dist-packages/pyftdi/i2c.py", line 766, in exchange
    self._do_prolog(i2caddress | self.BIT0)
  File "/usr/local/lib/python3.7/dist-packages/pyftdi/i2c.py", line 1014, in _do_prolog
    self._send_check_ack(cmd)
  File "/usr/local/lib/python3.7/dist-packages/pyftdi/i2c.py", line 1046, in _send_check_ack
    raise I2cNackError('NACK from slave')
pyftdi.i2c.I2cNackError: NACK from slave

I have tried the following - 
1. Lowering the frequency but the issue remains. 
2. I have tried the installation on both Raspbian Buster and Ubuntu 18.04 with the same issue. 
3. Swapped the FT232H board with the same issue. 
4. The sensor works fine when connected to an arduino. 

Has anyone came across such an issue and managed to solve it? 

Thank you.

eblot, 2020-03-31:
(edited initial report to fix markdown syntax)

eblot, 2020-03-31:
1. What pull-up resistor values do you use on SCL & SDA?
2. Check with `i2cscan.py` whether an I2C device is detected on the I2C bus
3. If you have a logic analyzer / scope, it would be great to see the FTDI signalling on AD0 and AD1.

sethpoh, 2020-03-31:
Hi there,

The sensor breakout board in question is here - https://www.adafruit.com/product/1980

1. It has 10k resistor pullup for the SCL and SDA.
2. The board uses I2C 7-bit address both on 0x29 and 0x28. I am able to detect it on 0x28. I have tried both addresses with the same result.
3. I will see whether I can obtain a trace of that.

Thank you.

eblot, 2020-03-31:
2. I meant, with PyFtdi's `i2cscan.py`. I'm not sure to understand what/how you tested this.

sethpoh, 2020-03-31:
Yes, I detected it on address 0x28 with the i2cscan.py.

eblot, 2020-03-31:
Ok, so it is weird, as it means PyFtdi is able to successfully read from the device.
Which PyFtdi version are you using?

sethpoh, 2020-03-31:
I am using 0.48.3. But I also tried a few earlier versions down to 0.40.0 with the same error message.

eblot, 2020-03-31:
So if the traceback is from 0.48.3, this means:
 - you can read from the device (`i2cscan.py`)
 - you can write to the device (your request which does not seem to trigger a NACK)
 - you cannot write-then-read in the same transaction, i.e. w/o a STOP between write and read.

I do not know the adafruit library, but why is it using the ``exchange`` method to read a register?

eblot, 2020-03-31:
Do you observe the same behaviour with:

    i2c = I2cController()
    i2c.configure('ftdi:///1')
    slave = i2c.get_port(0x29)
    devid = slave.read_from(0x12, 1)[0]
    print(hex(devid)) # should be 0x50

sethpoh, 2020-03-31:
Unfortunately I am not familar enough with the Adafruit_Blinka library to comment on the use of exchange method. More info can be found here if it's useful- https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/setup 

Yes I observed the same behaviour. When running the line  - 

`devid = slave.read_from(0x12, 1)[0]`

with the error - 
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/pyftdi/i2c.py", line 148, in read_from
    out=self._make_buffer(regaddr), readlen=readlen, relax=relax)
  File "/usr/local/lib/python3.7/dist-packages/pyftdi/i2c.py", line 766, in exchange
    self._do_prolog(i2caddress | self.BIT0)
  File "/usr/local/lib/python3.7/dist-packages/pyftdi/i2c.py", line 1014, in _do_prolog
    self._send_check_ack(cmd)
  File "/usr/local/lib/python3.7/dist-packages/pyftdi/i2c.py", line 1046, in _send_check_ack
    raise I2cNackError('NACK from slave')
pyftdi.i2c.I2cNackError: NACK from slave
```

eblot, 2020-03-31:
Well, I'll try to run new tests to check if there could be a regression I haven't caught.
Meanwhile, if you can get some capture of the SCL/SDA lines, it can also help.

sethpoh, 2020-04-02:
Hi @eblot,

I managed to identify the issue. I believe it was down to the cable length between the TSL2591 and the FT232H. Some observations I have are -

1. From the oscillscope trace, I can see that SCL voltage levels are not consistent/corrupted and sometimes below the logic high threshold.
2. Putting the TSL2591 right on the same breadboard as the FT232H solves the problem
3. I was running about 50 cm of cat 4 between them, shortening it by 10 to 15cm solves the problem.
4. After the cable legnth issue was solved, the FT232H was connected to an USB hub. that somehow also produces the NACK error. A direct connection to the Raspberry Pi I am running the script from solves the problem

A remaining question I have is, previously even with the longer cable, it is working with an Arduino. Before this I also tried lowering the FT232H frequency but that did not solve the issue. I wonder what are the plausible explanation for this?

I am closing the the issue here. Thank you for your help in diagnosing the issue.

eblot, 2020-04-02:
(Reopening till we are sure about the issue)

eblot, 2020-04-02:
I think the first thing to check would be the pull up resistors. 
10kΩ is likely too strong. Try with a 1.8KΩ or 2.2KΩ.

This is where a scope would help, as I suspect the signals on your I2C bus is likely to experience slow rising signals ([t = RC](https://en.wikipedia.org/wiki/RC_time_constant)). You should be able to use longer I2C bus wires with smaller resistors.

trigosot, 2020-05-05:
I am using cable USB/I2C converter C232HM-DDHSL-0 and had the same problem of I2cNackError.
The problem is that SCL rise at the same time than SDA falls before the START_BIT when the READ is performed. After set SCL = 1, it should wait `t_buf` before set SDA = 0.
The way I solved it is editing the file i2c.py modifying the ` def _do_prolog() ` function at line 1013, as follows:
```
def _do_prolog(self, i2caddress: int) -> None:
        if i2caddress is None:
            return
        self.log.debug('   prolog 0x%x', i2caddress >> 1)
        cmd = bytearray(self._idle * self._ck_delay)
```
The statement `* self._ck_delay` in the <cmd> gives the time set as `t_buf` defined in
```
'I2CTimings', 't_hd_sta t_su_sta t_su_sto t_buf'
...
I2C_100K = I2CTimings(4.0E-6, 4.7E-6, 4.0E-6, 4.7E-6)
I2C_400K = I2CTimings(0.6E-6, 0.6E-6, 0.6E-6, 1.3E-6)
I2C_1M = I2CTimings(0.26E-6, 0.26E-6, 0.26E-6, 0.5E-6)
...
ck_buf = self._compute_delay_cycles(timings.t_buf)
self._ck_idle = max(ck_su_sta, ck_buf)
self._ck_delay = ck_buf
```

eblot, 2020-05-06:
Ok, I'll have a look at this issue, thanks for the feedback.

trigosot, 2020-05-06:
> I am using cable USB/I2C converter C232HM-DDHSL-0 and had the same problem of I2cNackError.
> The problem is that SCL rise at the same time than SDA falls before the START_BIT when the READ is performed. After set SCL = 1, it should wait `t_buf` before set SDA = 0.

Evidence:
[image]
[image]

After:
> 1017        cmd = bytearray(self._idle * self._ck_delay)

[image]

eblot, 2020-05-08:
I do not get why SCL is low at the beginning, it is supposed to be high from the last stop bit.

Can you provide the full sequence you use to generate these traces? Are you using I2C+GPIO, or only I2C on this FTDI port?

trigosot, 2020-05-11:
Hi,
I am using only I2C on this FTDI, in fact when you use command read(), write() the SCL remains High from the last stop bit.
This issue happen the the commands: read_from(), exchange(). If you look deeper on this functions they perform first a write and a read after, the problem is that the stop bit from the write is not working properly affecting the Start Bit from the read. This Start Bit is what I shared in the last post.
Exchange:
* Write = Start bit + Slave Add + W + **Ack**  + DATA + **Ack** + (STOP BIT MISSING)
* Read = (Pseudo Start Bit) + Slave Add + R + **NAck** (ISSUE HAPPENS) 

If you check the exchange function you´´ll see:
```
def exchange(self, address: int,
                 out: Union[bytes, bytearray, Iterable[int]],
                 readlen: int = 0, relax: bool = True) -> bytes:
...
                    self._do_prolog(i2caddress)
                    self._do_write(out)  # After a Write there is a **self._do_prolog**
                    self._do_prolog(i2caddress | self.BIT0)
                    if readlen:
                        data = self._do_read(readlen)
                    do_epilog = relax
                    return data
...
```
Checking the function _do_prolog
```
 def _do_prolog(self, i2caddress: int) -> None:
        if i2caddress is None:
            return
        self.log.debug('   prolog 0x%x', i2caddress >> 1)
        cmd = bytearray(self._idle) # This **_idle** set the SCL but dosen't wait the t_buf time
        cmd.extend(self._start)
        cmd.extend(self._write_byte)
        cmd.append(i2caddress)
```
the way I correct this is adding **self._ck_delay**  to the command:
`1017        cmd = bytearray(self._idle * self._ck_delay)`

eblot, 2020-05-12:
Ok, I got it, I'll try to reproduce and fix this.

eblot, 2020-09-26:
Could you check that v0.52.0 fixes this long standing issue? Thanks.

eblot, 2021-01-11:
Should be fixed in v0.52.9. Feel free to re-open if not.

closed 2021-01-11 by eblot