eblot/pyftdi #180: Enable disable clockstretching

Potemkinkuliss, 2020-04-03:
Hey. I have a lot of fun working with this module. Thank you very much. It really works great on my setup.

One of my sensors requires clockstretching. The others Timeout if clockstretching it is enabled.
I am not sure why. Since to my understanding clockstretching is only affecting the clock if the slaves requires this.
Right now I `configure(..., clockstretching=True/False)` before and after reading from the sensor needing clockstretching. Since a method like `enable_adaptive_clock` is provided, I wonder if it would be possible to have something like `enable_clockstretching(bool)`.

eblot, 2020-04-03:
I guess it could be added.

However it is weird a non- clockstretching device cannot work if this mode is enabled.

Have you checked that one device on you bus does not pull down the SCL line when it is not supposed to do so?

Potemkinkuliss, 2020-04-07:
I finally found the answer. As expected this behaviour was caused by a bug in my implementation. Thanks for the quick reply. And thank you again for this great library!

closed 2020-04-07 by Potemkinkuliss

eblot, 2020-04-07:
Thanks for letting me know, and thanks for your support.