eblot/pyftdi #181: Problem regarding pyftdi and melexis 90621 sensor

AndreiUngureanu101, 2020-04-06:
Hello, I'm having trouble using the pyftdi library with an ft232h device and an melexis 90621 sensor:
The melexis 90621 sensor itself works perfectly when connected directly through the I2C pins to an arduino. However, I'm trying to connect a melexis 90621 through a ftdi 232 device and when I'm sending or receiving the same signals as I would using the arduino, I get a big pause in communication after some bytes, aroung 15ms. Now this big pause is the only difference when analyzing the signals using ftdi vs arduino and it seems to be the reason why the arduino version receives the correct data from the sensor and the ftdi receives static data, data that's not updated whatsoever. Thoughts?

eblot, 2020-04-06:
To perform an I2C read out, it takes many MPSSE commands (that is the commands of the FTDI protocol). When I say many, that's a lot of them: to read a single byte from a slave, 36 bytes are needed for the FTDI device, then a couple of other commands are required to read the buffered response.

The FTDI device uses two internal buffer/FIFO, whose size depends on the FTDI model. A FT232H has two FIFOs (TX, RX) of 1024 bytes. This means that the TX FIFO may contains 1024/36 = 28 read commands. 

As PyFtdi is mono-threaded, it creates a "read bytes from slave command", as many as it can fit into the FTDI FIFO, that is 28 sequences. The FTDI starts executing the read sequence as soon as it receives the first command bytes in its FIFO. PyFTDI then waits for the RX FIFO to be filled in with the I2C output data - that is 28 read bytes - retrieves them over the USB bus, and back to its client. At this point, it moves to the next chunk of commands to send to the FTDI device and so on.

This translates on the I2C bus as burst of 28 bytes, then long inactivity delays before PyFTDI has sent the next sequence of commands. This explains the waveform you are observing at a macro level.

The delay between burst is due to many parameters, such as the USB latency between micro frames and the whole SW stack (Python interpreter, PyFtdi, PyUSB, libusb, user/kernel border crossing, kernel USB stack,  USB host controller, ... back and forth for each transfer). It is also due to several settings in FTDI (latency for example), USB buffer size, ...

I think there is a lot of room for improvement, at least for I2C read. I2C write requests are fully inefficient, as for each *byte* written to the slave PyFtdi needs to retrieve the ACK/NACK from the slave, which ends up with a crazy protocol overhead - i.e. FTDI sucks at I2C :P

If I decompose the I2C read loop, this gives on my machine:
* read command chunk generation: < 1µs
* send chunk to FTDI: 30 ms
* read response from FTDI: 45 ms

That's far worse than your own figures (75 ms between bursts on my Saleae, vs. 15 ms with your Saleae capture).

It should be theoretically possible not to wait for the read response from the FTDI to emit the next read command chunk, but it's quite a piece of work given the current architecture. 

Tweaking some parameters such as the read timeout, FTDI latency, etc. may also improve the overall figures which much less work.

I'll keep all this on my TODO list.

However, if you really need some kind of "real time" I2C readout from a sensor, I would not recommend PyFTDI for now, and I'm not even sure it can be really reliable with an FTDI bridge given the way I2C needs to be implemented. You can find microcontrollers that would do a much better job at interfacing a host with an I2C slave - even an Arduino does a much better job.

AndreiUngureanu101, 2020-04-07:
Thank you for the detailed explanation! It's very helpful!

eblot, 2020-04-08:
Ok, a quick and dirty hack shows that it is possible to improve quite a lot I2C read transfers. It is too early to even release a patch, but I'll try to build something useful this week.

eblot, 2020-04-08:
Can you give a try to the (i2c_streaming)[https://github.com/eblot/pyftdi/tree/i2c_streaming] branch?

It is a quick experiment to optimise I2C transfer bandwidth and most importantly reduce I2C bus stalls.

Using an I2C EEPROM as a data source, reading out 4096 bytes, on FT232H:
 -w/o optimisation: 2350 ms read time (1 KB/s), about 15 ms stall every 28 bytes
 - w/ optimisation: < 200 ms read time (20KiB/s), stall varies from 95 µs to 2ms, the average should be around 250 µs, every 14 bytes.

I'm not sure it works with all setups, but I'd be interested in your feedback.

AndreiUngureanu101, 2020-04-08:
You're right, the stall is better now, around 0.15 ms in my tests: 
[image]

eblot, 2020-04-08:
Does it improve the reliability with your sensor ?

AndreiUngureanu101, 2020-04-08:
Yes, it does, but that's not the sole issue. I need to send commands to my sensor in order to write some configuration values and the commands I send don't seem to have any effect on the values that I'm supposed to change. And I'm guessing this issue is due to these delays. I found the libmpsse library that does work in my specific case.

eblot, 2020-04-09:
Ok, too bad, sorry about that. Could you post a Saleae capture of what libmpsse achieves? I do not think it is possible to be as efficient with Python, but there's still room for improvement :-)
Thanks.

AndreiUngureanu101, 2020-04-09:
Well libmpsse uses swig to load in C code, so my understanding is that this is what makes it fast

[image]

eblot, 2020-04-09:
Definitely, a C implementation do go much faster than a Python one. SWIG is some kind of legacy way to access native code, but it is does not change the performances.

This is still weird that 1ms delay between commands change the behaviour of a sensor, especially in the configuration stage... Maybe there is another hidden issue. I do not have this sensor, so I'm afraid I cannot investigate more for now.

AndreiUngureanu101, 2020-04-09:
Yes, no problem, I understand. I'll figure something out using the libmpsse library. Thanks again!

eblot, 2020-04-09:
Thanks.
If you come across any useful info, please let me know.

I keep this ticket open till I can deliver this updated version with shorter stalls.

AndreiUngureanu101, 2020-04-09:
Ok, I'll let you know. For now libmpsse seems to work fine with the exception that the real clock rate is 2/3 out of the clock rate I'm setting (100kHz). The null clock takes twice as long as the one clock and together are 2/3 out of the desired clock rate.

eblot, 2020-04-09:
I think I can tell you why :-)

To make I2C working, FTDI needs to enable what they call a "3 phase clock". PyFtdi implements this feature. The issue with this mode is that it requires 3 clock phase for each clock period, hence the 2:3 ratio. PyFtdi hides this setting from the user, and silently tweaks the specified clock to match what the users expects. `libmpsse` seems not to do it.

AndreiUngureanu101, 2020-04-09:
I understand. But does this influence the correct functioning of the setup? Is everything working as it should at this new (2/3)*setup frequency. Up until now it seems it does but I'm guessing you know if something might go wrong.

eblot, 2020-04-09:
I do not think it changes anything:
* pyftdi changes the submitted frequency so that it is 3:2 the one provided when I2C mode is activated
* libmpsse expects the user to fix the frequency value himself. As D2XX expects the user to multiply the frequency by 16 when he wants to use the bitbang mode with a non-H series chip, or by 5 with a H-series chip, and many other kludges :)

AndreiUngureanu101, 2020-04-09:
I see, that's good to hear :)

eblot, 2020-04-28:
I've delivered the optimised I2C read version as v0.51.0.
I'm not sure I can do better for now, at least not with a dedicated HW slave to test.

AndreiUngureanu101, 2020-04-28:
Thanks, I'll test it out!

eblot, 2020-04-28:
There is nothing new since https://github.com/eblot/pyftdi/issues/181#issuecomment-611081709 I just delivered this work to the master branch.