eblot/pyftdi #182: Ftdi.list_devices() crashing on NotImplementedError unnecessarily

np-8, 2020-04-09:
#### Configuration
- Windows 10
- Python 3.7.7 65-bit
- libusb-1.0.dll

### Problem description
For some reason the `list_devices()` function was not working. Probably the error is related to some USB device I have connected, but the code to reproduce the error was
```python
from pyftdi.ftdi import Ftdi
Ftdi.list_devices()
```
which caused (Error on `pyusb` library side)
```python
~\myproj\venv\lib\site-packages\usb\backend\libusb1.py in _check(ret)
    591     if ret < 0:
    592         if ret == LIBUSB_ERROR_NOT_SUPPORTED:
--> 593             raise NotImplementedError(_strerror(ret))
    594         else:
    595             raise USBError(_strerror(ret), ret, _libusb_errno[ret])

NotImplementedError: Operation not supported or unimplemented on this platform
```

### Proposed workaround
- Skip the `NotImplementedError` , i.e. in `usbtools.py` replace 
```python
        try:
            if cls.UsbApi == 2:
                return usb_get_string(device, stridx)
            return usb_get_string(device, 64, stridx)
        except UnicodeDecodeError:
            # do not abort if EEPROM data is somewhat incoherent
            return ''
```
with
```python
        try:
            if cls.UsbApi == 2:
                return usb_get_string(device, stridx)
            return usb_get_string(device, 64, stridx)
        except (UnicodeDecodeError, NotImplementedError):
            # do not abort if EEPROM data is somewhat incoherent
            return ''
```

eblot, 2020-04-10:
Hi,

I'd like to understand the exact error before silently trashing a potentially legit error. The existing exception handler is dedicated to avoid discarding a working device whose EEPROM content has been malformed, which may easily occur.

However, trashing a `LIBUSB_ERROR_NOT_SUPPORTED` is another beast, and may lead to other issues. Did you run libusb in debug mode to get more details about the actual error?

np-8, 2020-04-10:
Hi,

No I did not run libusb in debug mode. Is that something that can be easily done? I have to say that I needed to use a little hack to get my libusb-1.0.dll loaded (I have not installed anything, just copied the dll from the libusb repository), so what I did was:

```python
from pyftdi.usbtools import UsbTools
UsbTools.BACKENDS = UsbTools.BACKENDS + ('mypackage.mymodule', )
```
where `mypackage.mymodule` is a python module with a `get_backend`-function that returns the path of the `libusb-1.0.dll`. 

Anyway, I was just happy that I got the list of FTDI devices visible. After that I found out that maybe using the [ftd2xx](https://github.com/snmishra/ftd2xx) package was actually better suited for my needs since I just wanted to read serials, Set some RTS flags, and get some CTS-flags. 

I have the device "not supported by libusb" on my desktop now so I could try to check out the debug logs, if it helps someone. When listing the FTDI devices, I did not really care if the libusb drivers support the device or not. I think that maybe giving a some kind of "BackendNotSupportedWarning" would be then ok? In any case, if the backend is not supported and I would try to do something else than just read the serial number, I would expect an Exception to be raised anyway.

eblot, 2020-04-10:
> No I did not run libusb in debug mode. Is that something that can be easily done?

On Linux/macOS, it is quite simple, yes: you just need to export `LIBUSB_DEBUG` environment variable and set the trace level, see http://libusb.sourceforge.net/api-1.0/. On Windows, I have really no idea.

> I have the device "not supported by libusb" on my desktop

Is it the device that is not supported, or a specific call?
This is weird, as only configured devices through Zadig should appear in the listed device, so a device that you have not explicitly added should not get in the way.

No problem if you prefer to use ftd2xx. 

Thanks for the update.

eblot, 2020-04-10:
Reading once more your last post: does it mean you bypassed the installation instructions, and you have not gone through the Zadig setup?

Maybe that would be the first steps to take before starting to hack the code :-)

eblot, 2020-04-16:
It seems the error comes from an invalid setup of USB library.

If you can reproduce the error with a proper Zadig config, please reopen this ticket.

closed 2020-04-16 by eblot