eblot/pyftdi #185: GpioMpsseController set_direction() problem

enesbcs, 2020-04-19:
I experienced a little odd results with set_direction(). When i use it with GpioAsyncController like this:

```
import time
from pyftdi.gpio import GpioAsyncController

gpio = GpioAsyncController()
gpio.configure('ftdi://ftdi:232h:2:9/1', direction=32)
gpio.set_direction(32,32)
print("AD5 low")
gpio.write(0)
time.sleep(10)
print("AD5 high")
gpio.write(32)
time.sleep(10)
gpio.close()
```

The results are OK AD5 go to low, then high.
But when i try the same with GpioMpsseController:

```
import time
from pyftdi.gpio import GpioMpsseController

gpio = GpioMpsseController()
gpio.configure('ftdi://ftdi:232h:2:9/1', direction=32, frequency=6.0E6)
gpio.set_direction(32,32)
print("AD5 low")
gpio.write(0)
time.sleep(10)
print("AD5 high")
gpio.write(32)
time.sleep(10)
gpio.close()
```

The AD5 output stays at low. After the set_direction() command, the write() functions fails to physically change the pin states from some reason. Maybe set_direction() is not adapted as mpsse function? There are no errors on the console.

eblot, 2020-04-20:
There is indeed an issue with `set_direction`.

I tried with both virtual and real hardware, and it does fail.

Meanwhile, did you try w/o the `set_direction` call, as it is already defined in the `configure` method?

enesbcs, 2020-04-20:
Thank you, yes, it works with the preconfigured value. But in this way i have to close and reopen the connection if i want to modify a pin state.

eblot, 2020-04-20:
Ok, got it. If you enable the debug traces, you can see that bitbang mode is called on `set_direction` from MPSSE mode, which is not expected at all :-) Working on a fix.

closed 2020-04-20 by eblot

enesbcs, 2020-04-20:
For MPSSE I2C & SPI their own _set_gpio_direction() function works fine.

eblot, 2020-04-21:
Delivered in v0.50.2

commit c032f2bd (2020-04-20): Fix #185. `set_direction` was only valid for `GpioAsyncController`.