eblot/pyftdi #188: I2C scanner

enesbcs, 2020-05-05:
I've tested the i2c scanner (pyftdi/bin/i2cscan.py) and it works ok with may few devices i've tried.

But the original i2cdetect works somehow this:
```
for (i = 0; i < 128; i += 16) {
 for (j = 0; j < 16; j++) {
 if ((i+j >= 0x30 && i+j <= 0x37)
  || (i+j >= 0x50 && i+j <= 0x5F))
	res = i2c_smbus_read_byte(file);
 else
	res = i2c_smbus_write_quick(file,
	      I2C_SMBUS_WRITE);
}}
```
Are there any way to mimic smbus_write_quick() method with pyftdi? As read may lock SMBus on various write-only chips.

And one more question: when scanning i2c devices warnings popping up on the console, could we silence them somehow?

```
...
WARNING: NACK @ 0x6a
WARNING: NACK @ 0x6b
WARNING: NACK @ 0x6c
WARNING: NACK @ 0x6d
WARNING: NACK @ 0x6e
WARNING: NACK @ 0x6f
WARNING: NACK @ 0x70
...

```

eblot, 2020-05-06:
I do not know the specs of SMBus, what is a write_quick? A simple write request with no data? In this case, I guess it could be easily implemented.

The warnings come from the log, it can also be improved. In tnis very specific use case, no warning should be emitted.

I'll have a look at all this as soon as I get some spare time, I hope by the end of the week.

enesbcs, 2020-05-06:
Thank you! According to kernel.org the Quick write command "sends a single bit to the device, at the place of the Rd/Wr bit."
https://www.kernel.org/doc/Documentation/i2c/smbus-protocol
Essentially it looks like writing without data.

eblot, 2020-05-08:
> And one more question: when scanning i2c devices warnings popping up on the console, could we silence them somehow?

I'm not able to reproduce this issue.
`i2cscan` explicitly changes the log level to ERROR (@ line 56), so the warning trace should not be emitted....

eblot, 2020-05-08:
> The SMBus was defined by Intel
LoL
I should have bet on this: when there's something weird and clunky, 9 chances out of 10 Intel has been involved in the design...

eblot, 2020-05-08:
SMBbus scanning should now be the default scanning mode.

If your warning message issue persists, please open a new ticket, thanks.

commit 00235515 (2020-05-08): Fix #188
Merge branch 'issue_188'
closed 2020-05-08 by eblot