eblot/pyftdi #189: I2C + GPIO Writes resetting BD Bus

dua564, 2020-05-07:
Hi eblot,

I'm using as FT2232H chip with interface 2 configured as I2C.  This interface is being used as an I2C Controller to a MAX5825 device and has 12 GPIOs. BC7-0, BD7-4

I'm noticing that whenver I perform GPIO writes to the bus, if i perform an I2C read right after the write, the GPIOs on BDBus lose state and float to 2.5V, while the GPIOs on BCBus maintain state,

As an example, I have configured my code as following:

test_max5825 = MAX5825(url=url_i2c, frequency=100E3, address=0x12)
gpio = GPIO_Buffers(url=url_i2c, frequency=100E3)

#where gpio.open() contains
# self.i2c.configure(url=self.url)
#  self.slave = self.i2c.get_port(address=0x12) #Fake I2C Address for GPIO Functionality
#  self.io = self.i2c.get_gpio()
#  self.io.set_direction(0xFFF0, 0xFFF0)

test_max5825.open()
gpio.open()
   
# all output set lo
 gpio.io.write(0x000)  --> I see all lows
# all output set high
 gpio.io.write(0xFFF0)    --> i see all Highs 
# all output set high, apply direction mask
 gpio.io.write(0xFFF0 & gpio.direction)  --> I see all highs 

test_max5825.read_reg(address=0xc0, verbose=True) --> self.slave.read_from(regaddr=address, readlen=2)

#After peforming this command, all signals on BD bus float to 2.5V instead of 3.3V. BC Bus is held at 3.3V

Any ideas on what may be happening here?

-Nik

eblot, 2020-05-08:
What is `GPIO_Buffers`?

Can you provide the call you make to PyFtdi APIs? Thanks.

dua564, 2020-05-12:
Class is listed below. URL and port is both same.  However, am I able to instantiate a I2C Controller twice as below for each device or should I2C controller be only defined once and each device would get called as a slave for a single controller?

test_max5825 = MAX5825(url=url_i2c, frequency=100E3, address=0x12)
gpio = GPIO_Buffers(url=url_i2c, frequency=100E3)

test_max5825.open()
gpio.open()

class GPIO_Buffers:

    def __init__(self, frequency, url):

        self.i2c = I2cController()
        self.frequency = frequency
        self.slave = None
        self.io = None
        self.url = url
        self.pins = 0xFFF0 #Pins C7 to D4 are being used as GPIOS
        self.direction = 0xFFF0       
        self.state = 0x0000 

    def open(self):
        """Open an I2C connection to a slave"""

        self.i2c.configure(url=self.url)
        self.slave = self.i2c.get_port(address=0x12) #Fake I2C Address for GPIO Functionality
        self.io = self.i2c.get_gpio()
        self.io.set_direction(self.pins, self.direction)

class MAX5825:

    def __init__(self, frequency, url, address=0x12):

        self.i2c = I2cController()
        self.frequency = frequency
        self.slave = None
        self.io = None
        self.url = url
        self.pins = 0xFFF0  # Pins C7 to D4 are being used as GPIOS     
        self.direction = 0xFFF0
        self.state = 0x0000
               

    def open(self):
        """Open an I2C connection to a slave"""

        self.i2c.configure(url=self.url)
        self.slave = self.i2c.get_port(address=self.address)
        self.io = self.i2c.get_gpio()
        self.io.set_direction(self.pins, self.direction)

      def read_reg(self, address, verbose=False):
        """ Read byte data from single register at address
        address = hex value in byte (eg: 0xAB)
        returns: int value of data at address "
        """

        data = self.slave.read_from(regaddr=address, readlen=2)
        hex_data = str('0x' + hexlify(data).decode())

        if verbose is True:
            print('Register Value at address ' + hex(address) + ': ' + hex_data)
            print('')

        return int(hex_data, 16)

eblot, 2020-05-13:
If I get it right, you instantiate TWO I2cController (which do not know each other) to drive the very same port of the same device, since the URL is the same.

You cannot use 2 concurrents controller for the same port. Doing so, each time a FTDI request is performed, it overrides the request made by the other controller, i.e. this is a race condition.

You want to instantiate *one* I2cController, retrieve an I2cPort from it to drive the MAX5825, and retrieve the GpioPort from the same controller to drive the GPIOs. This way, the controller is able to cope with concurrent requests.

dua564, 2020-05-14:
Ahh Got it!  Seems like this old approach worked with pyftdi 0.29 as I have many slaves on both SPI / I2C bus instantiated like this without issues but doesn't seem work with latest 0.5. 

Things are working now, however, with one I2c controller so thanks for that! I'm guessing if I have a second SPI controller with GPIO, I would have to instantiate it the same way as I2C+GPIO correct?
 As in one single self.io for GPIO and separate self.slaves for all the CS.

eblot, 2020-05-16:
You need to instanciate one and only one controller per port, this means:
 * 1 controller on FT232H
 * up to 2 controllers on FT2232H
 * up to 4 controllers on FT4232H, although MPSSE is only implemented in HW for the two first port in this device.

A *controller* is any of I2cController, SpiController, Gpio*Controller.

If it has worked in some previous release, it was an unexpected side effect,* i.e.* you were very lucky as it has never been designed to :-)

closed 2020-05-16 by eblot

stevessp, 2023-02-07:
This also solves my problem. I had two threads each with their own controller. Now that they share one we're good.