eblot/pyftdi #190: Bitbang seems broken after packet update

RyanHope, 2020-05-21:
I updated one of my projects to use the latest verison of pyftdi and bitbang seems to be broke or there was a change I am not aware of. Previously this code was working:

```
dev.set_bitmode(1, 0x01)
dev._write(b'\x00')
time.sleep(.070)
dev._write(b'\x01')
dev.set_bitmode(0, 0x00)
sleep(.120)
dev._read()
```

Basically this code sent a init signal, serial held low for 70ms then high for 120ms. This part of the code no longer seems to have the desired effect of waking up the device I am trying to talk to. Not sure what the fix is.

closed 2020-05-22 by RyanHope

eblot, 2020-05-22:
Wrong project?

RyanHope, 2020-05-22:
No not wrong project but I think the issue had to do with my projects build scripts reinstalling wrong deps. I saw that there is a change for Ftdi.BitMode and I ported my code to use that and I was still having issues. Its all resolved now.

eblot, 2020-05-22:
Ok.  It seems you are using the internal APIs (`._read`, `._write`), so there is no guarantee it stays stable from one version to another.
`Gpio*Controller`s should provide safer APIs