eblot/pyftdi #191: Can not write to eeprom on 4232h

Renha, 2020-05-23:
Could not write to empty eeprom on 4232h, always get the error and nothing is written.
Checked versions 0.49.0 and 0.51.2.

eblot, 2020-05-23:
Could you provide some details?

- what API or command did you use and how,
- what error message did you get, if any
- have you tried with another device?
- anything useful that can help understand and reproduce the issue, as “it does not work” is a bit difficult to troubleshoot :-)

Thanks.

Renha, 2020-05-24:
Hey! There's chance of the problems on the schematics, I'll update the case accordingly when it will be confirmed or not.

Renha, 2020-07-02:
Oh, sorry! That was hardware issue! Thank you

closed 2020-07-02 by Renha

eblot, 2020-07-02:
Thanks for the feedback.