eblot/pyftdi #192: Cant poll available devices more than once on Windows / USBDK

RyanHope, 2020-05-26:
Since official hotplug in libusb seems broken in Windows, the only option is to manually poll for new/removed devices. Here is some very simple code:

https://gist.github.com/RyanHope/a24dd0cf29753f83ffaa8039a62c163e

On Linux this works without a problem but on Windows it crashes:

```
<pyftdi.ftdi.Ftdi object at 0x02D4CBF0>
<pyftdi.ftdi.Ftdi object at 0x02D33BF0>
<pyftdi.ftdi.Ftdi object at 0x02D4CBF0>
Traceback (most recent call last):
  File ".\test2.py", line 43, in <module>
    d.open_from_device(iface.device, iface.index + 1)
  File "C:\Python37-32\lib\site-packages\pyftdi\ftdi.py", line 569, in open_from_device
    config = self._usb_dev.get_active_configuration()
  File "C:\Python37-32\lib\site-packages\usb\core.py", line 909, in get_active_configuration
    return self._ctx.get_active_configuration(self)
  File "C:\Python37-32\lib\site-packages\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Python37-32\lib\site-packages\usb\core.py", line 247, in get_active_configuration
    self.managed_open()
  File "C:\Python37-32\lib\site-packages\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Python37-32\lib\site-packages\usb\core.py", line 131, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "C:\Python37-32\lib\site-packages\usb\backend\libusb1.py", line 905, in open_device
    return _DeviceHandle(dev)
  File "C:\Python37-32\lib\site-packages\usb\backend\libusb1.py", line 715, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "C:\Python37-32\lib\site-packages\usb\backend\libusb1.py", line 660, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno None] Other error
```

eblot, 2020-06-07:
Well, Windows is not officially supported (https://github.com/eblot/pyftdi#supported-host-oses) because I simply deeply hate this OS, and it is already some kind of miracle it sometimes works with Zadig...

It looks like it is more an issue with `pyusb` than `pyftdi`, but patches are always welcomed.

mpratt14, 2020-06-12:
@RyanHope Well you said it yourself...

> On Linux and Windows with zadig/libusbK this works without a problem, but on Windows with USBDK it crashes:

Sounds like you should open an issue with the USBDK developers, instead of here

eblot, 2020-06-27:
> Sounds like you should open an issue with the USBDK developers, instead of here

You are right @mpratt14 
@RyanHope Please use Zadig w/ Windows, or provide a patch but I think this is not directly related to PyFtdi but to the lower APIs (PyUSB or the native USB lib)

closed 2020-06-27 by eblot