eblot/pyftdi #193: How to reduce the SCL delay between Address and Data when use Write CMD?

zhen-pan, 2020-05-31:
Hello, 
I am using FTDI to control a simple I2C slave which does not have register address, it is 1 single register device(TCA9548A,  I2C MUX from TI)
My code:
slave = i2c.get_port(0x70)
slave.exchange([0x80],1) 

So basically what I need to do is to to start a write with address 0x70 and send 0x80 after that, but right now I am see a ~120us delay between address command and data command 0x80

How would I reduce the CLK delay between address and data? 
Thanks

[image]

eblot, 2020-06-01:
I do not think it can be reduced much:

I2C protocol is not handled by FTDI HW, but by the host - this is the way FT*232 devices work.

When the host sends the first byte (start, slave address), it needs to read back the acknowledgement status bit from the slave before resuming - to check whether the I2C slave is on the bus and ready to receive the next byte. Once this ACK bit is received, it may resume with the next byte to send.

This sequence goes over the USB bus and USB stack, libusb, Python interpreter, pyusb, pyftdi. The write-read-write roundtrip is likely not to reduced much with write sequences...

eblot, 2020-06-01:
Conversely, read sequences are now optimised as the host does not have to check for ACK bit, since it is in charge of generating them, so it pushed then to the FTDI device w/o having to go through a round trip on each byte.

zhen-pan, 2020-06-01:
> I do not think it can be reduced much:
>
> I2C protocol is not handled by FTDI HW, but by the host - this is the way FT*232 devices work.
>
> When the host sends the first byte (start, slave address), it [...]

Got it, thanks for reply Emmanuel

eblot, 2020-06-07:
Closing as it is a HW limitation, with no known workaround.

closed 2020-06-07 by eblot