eblot/pyftdi #194: JTAG shift_register does not work

markmelvin, 2020-06-04:
I am trying to use pyFTDI with JTAG and I believe the current version of JtagEngine is broken.  

Simple writes and reads with `write_ir` and `read_dr` do seem to work but I want to write bits to TDI while reading bits off TDO, so I am using `shift_register`.  There seems to be a couple issues:

1) Simply using `JtagEngine.shift_register` doesn't work at all because it doesn't pass through the `use_last` argument so there is no way to specify this so that TMS transitions on the last bit. Thus, as far as I see it - you end up sending one bit too many - which causes data problems going out.
2) Using `JtagEngine.jtag.shift_register()` I can pass the argument `use_last`, which appears to properly transition the TAP state machine aligned with the data, but it ends up not shifting the data properly coming off the JTAG TDO pin as the last data bit is sent in unison with the TMS signal set to 1 bit the last incoming bit is never read - thus the incoming data is wrong.

Am I using this incorrectly?  It seems the simultaneously reading and writing is broken.

eblot, 2020-06-07:
Unfortunately I have used JTAG for nearly 10 years, so it may be broken.

I've wondered several times whether I should remove its support from `pyftdi`... Patches are always welcome - see other tickets related to JTAG if it may help.

Sorry about that.

markmelvin, 2020-06-07:
I have a working solution that I can turn into a patch this week. I wouldn't remove it. It is a clever implementation!

eblot, 2020-06-08:
Great, and thanks.

eblot, 2020-06-18:
Please create a PR if you have a fix for this issue. Thanks

markmelvin, 2020-06-18:
Look up. I already did...

eblot, 2020-06-27:
> Look up. I already did...
Very sorry, I missed it.

eblot, 2020-09-26:
Merged. Thanks for the investigation and the patch. Sorry for the delay.

closed 2020-09-26 by eblot