eblot/pyftdi #195: Error running pyterm.py

langestefan, 2020-06-08:
Hi,

While running pyterm.py I get the following error:

`C:\Users\langestefan\gits\pyftdi-master\pyftdi-master\pyftdi\bin>python pyterm.py ftdi://ftdi:2232/1 
Traceback (most recent call last):
  File "pyterm.py", line 56, in <module>
    from pyftdi.misc import to_bps, add_custom_devices
ImportError: cannot import name 'add_custom_devices' from 'pyftdi.misc' (C:\Python38\lib\site-packages\pyftdi\misc.py)`

I am a bit confused by the description of pyterm: [https://eblot.github.io/pyftdi/api/uart.html#mini-serial-terminal](https://eblot.github.io/pyftdi/api/uart.html#mini-serial-terminal)

I am using windows so the commands in the description don't work for me. Do I have to set the PYTHONPATH variable?

Any help is appreciated!

langestefan, 2020-06-08:
Updating to the latest version got rid of that error. Now it fails here:

`Traceback (most recent call last):
  File "pyterm.py", line 460, in <module>
    main()
  File "pyterm.py", line 441, in main
    init_term(args.fullmode)
AttributeError: 'Namespace' object has no attribute 'fullmode'`

Not sure what to do now.

eblot, 2020-06-08:
Sorry, I barely test with M$ Windows.

Can you pull the latest master version and give it a try?

langestefan, 2020-06-08:
@eblot 

I did:

```
`C:\Users\320028010.CODE1>cd C:\Python38\Scripts

C:\Python38\Scripts>python pyterm.py ftdi:///?
Traceback (most recent call last):
  File "pyterm.py", line 460, in <module>
    main()
  File "pyterm.py", line 441, in main
    init_term(args.fullmode)
AttributeError: 'Namespace' object has no attribute 'fullmode'

C:\Python38\Scripts>pip3 list
Package                        Version
------------------------------ ---------
....
pyftdi                         0.51.2
....
```

Everything else seems to run perfectly fine. Thanks for the help!

eblot, 2020-06-08:
I think you pulled but did not reinstall:

\Scripts is not part of pyftdi, it is the destination directory for default script installation.

Either run from git directory, using `set PYTHONPATH=...` or run `setup.py` as you once did.

eblot, 2020-06-08:
Line 441 is now 

    init_term(full_mode)

langestefan, 2020-06-08:
You are completely right. I somehow messed up my installation. Pulled from master branch and now everything is working!

Thanks a lot for your hard work. Can I buy you a coffee somewhere?

closed 2020-06-08 by langestefan

eblot, 2020-06-09:
> Thanks a lot for your hard work.
Thanks for the feedback.

> Can I buy you a coffee somewhere?
I do not think, but thanks :-)