eblot/pyftdi #200: GPIO sample code bugs

newAM, 2020-06-18:
There are some syntax errors in the sample code for GPIOs, there is mismatched braces and unclosed quotes here.
```python
gpio = GpioSyncController()
gpio.configure('ftdi:///1, direction=0x0F, frequency=1e6)
outs = bytes(range(16})
ins = gpio.exchange(outs)
# ins contains as many bytes as outs
gpio.close()
```

Additionally the CBUS GPIO sample uses `return` without being in a function, and `self` without being in a class.

eblot, 2020-06-18:
Sorry about that. Thanks for the fix.

newAM, 2020-06-18:
No problem!  Thank you for this great library!

richey-v, 2020-06-18:
(I realize this issue is closed, but wanted to add...)

Have  you, @eblot, considered using [doctest](https://docs.python.org/3/library/doctest.html) on the sample code?

eblot, 2020-07-01:
@richey-v it might be worth considering this, yes, so this could be checked with unit tests. Thanks for the suggestion.