eblot/pyftdi #203: Possible I2C timing bug

akbyrd, 2020-06-20:
I noticed something that looks like a copy-paste bug in i2c.py. I2cController defines 3 possible timings, but always using the same one: I2C_100K

Here's the bit that looks off
https://github.com/eblot/pyftdi/blob/29174d3143a64ef0b0d603f6068d947f642d4164/pyftdi/i2c.py#L461-L467

And the timing definitions. Only 100K is referenced.
https://github.com/eblot/pyftdi/blob/29174d3143a64ef0b0d603f6068d947f642d4164/pyftdi/i2c.py#L390-L392

Looks like it's been this way since the original commit
https://github.com/eblot/pyftdi/commit/ef431724ca29438791b0aa6d9e27a808258bc3cf#diff-57181eb17d6e062e16e0a0befdbed5d9

eblot, 2020-06-22:
You are right, I fully forgot this point: I wanted to test the accuracy of the generated signals with a digital scope - because FTDI never behaves as you might think it would, then moved to another topic and forgot this one.

eblot, 2020-12-31:
The fix has been delivered in d4c8a1aeb51bfb85de54c77163a5f53c3d0f6d2c (v0.52.5)

closed 2020-12-31 by eblot