eblot/pyftdi #205: Stop violates I2C timing

DPM-PMC, 2020-06-25:
Current design has data idle high until stop is sent.  When stop is sent CLK immediately goes high and SDA immediately goes low.  However, SDA should not change until after the stop hold time timing spec.
[image]
Most parts this does not appear to be an issue.  However, the NXP PCA9848 I2C switch will not recognize it as a stop.

Recommend changing _stop() to start with a _clk_lo_data_lo to ensure that SDA is low before posedge of CLK.
[image]

This ensures SDA is low for the full stop_setup period.
[image]

eblot, 2020-06-27:
I think it is a dup of #179.  
Unfortunately I lack time at the moment to create an I2C setup and reproduce this, but it is a major issue to address, I agree.

naushir, 2020-08-27:
I've run into exactly the same issue with a TCA9548A I2C switch device attached to a FTDI2232H.

The above change has fixed the SDA transition violation for me as well.  @eblot, would you like me to create a PR for this fix?

eblot, 2020-08-27:
Yes please do. I really have not time to work on PyFtdi at the moment, but this issue needs to be addressed. Thanks.

naushir, 2020-09-01:
> @naushir Yes please do. I really have not time to work on PyFtdi at the moment, but this issue needs to be addressed. Thanks.

Done.  See PR https://github.com/eblot/pyftdi/pull/217

daveismith, 2020-09-23:
We were seeing the same issue on some boards with a MCP23017, though strangely only one of them since we've got two on board. I'm guessing it's because of trace length. As soon as we applied this fix, the device started to work reliably.

commit 62491f51 (2020-09-26): Merge pull request #217 from naushir/master

Fix #205 - I2C stop violation
commit cecc7e32 (2020-09-26): Fix #205 - I2C stop violation

Delay the SCL transition to correctly catch the SDA low->high transition.
closed 2020-09-26 by eblot