eblot/pyftdi #206: turn led on using  adafruit ft232h

glennford49, 2020-06-26:
how to set as HIGH on gpio pins of this ft232h device, i found it hard to just set as HIGH status of pins, unlike arduino gpio

mpratt14, 2020-06-26:
Arduino syntax is designed to be as easy as possible for users, since it is an educational device in many ways. This project is much more advanced.

See:
https://eblot.github.io/pyftdi/api/gpio.html
https://github.com/eblot/pyftdi/blob/master/pyftdi/tests/gpio.py

You need to:

1. Open the interface with open_from_url()
2.  Set direction and frequency of all 8 pins with configure()
3. Read or write 0x00 or 0xff

if you want easier syntax, I recommend installing Adafruit Blinka as a wrapper

https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/setup

I haven't written code using this library alone, but with Blinka it's pretty simple:

```
import board as FTDI
import digitalio as GPIO

CS_pin = FTDI.D7
Baudrate = 25000000

CS = GPIO.DigitalInOut(CS_pin)
CS.direction = GPIO.Direction.OUTPUT

while not FT232H.try_lock(): pass

FT232H.configure(baudrate=Baudrate)

#set pin low
CS.value = False

#set pin high
CS.value = True
```

eblot, 2020-06-27:
Thanks @mpratt14.

glennford49, 2020-06-27:
Thanks, ive setup blinka-adafruit , its now working

closed 2020-06-27 by glennford49