eblot/pyftdi #208: pyftdi.ftdi.FtdiError: UsbError: [Errno 16] Resource busy

glennford49, 2020-06-29:
Error when running 2 python interpreter, first.py is for setting high of pin, when  second.py running on different python interpreter for reading pin state, it throws an error,  is there any way to run this 2 python files with my ftdi adapter? Device is ft232h

mpratt14, 2020-06-29:
the FT232h only has 1 virtual interface. What you're suggesting is much more difficult than simply combining the two python files.

glennford49, 2020-06-29:
Any device you can suggest that has virtual interface greater than 1 ?

mpratt14, 2020-06-30:
See in the docs:
https://eblot.github.io/pyftdi/urlscheme.html

interface: the interface of FTDI device, starting from 1

1 for 230x and 232* devices,

1 or 2 for 2232* devices,

1, 2, 3 or 4 for 4232* devices

closed 2020-06-30 by glennford49