eblot/pyftdi #212: FT232R - CBUS GPIO output only switches for about 2ms

mani9876, 2020-07-24:
Hello,

I would like to use the CBUS feature of pyftfi, to switch the 4 GPIOs of an FT232R.

I programmed the EEPROM accordingly to set CBUS0 - CBUS3 in GPIO mode.

I set all the directions to output and set all GPIOs to high, but the pins are only set high for about 2ms (measured with oscilloscope).

Have anyone an idea, why this happens?

This is my code:

`import time`
`from pyftdi.ftdi import Ftdi`
`from pyftdi.eeprom import FtdiEeprom`
`ftdi = Ftdi()`
`ftdi.open_from_url('ftdi://ftdi:232:AR4SOIOR/1')`
`if ftdi.has_cbus:`
`        print("has cbus")`
`eeprom = FtdiEeprom()`
`eeprom.connect(ftdi)`
`print(eeprom.cbus_mask)`
`ftdi.set_cbus_direction(0b1111,0b1111)`
`ftdi.set_cbus_gpio(0b1111)`
`time.sleep(10)`

Thanks in advance!
Manuel

eblot, 2020-07-27:
This is weird because AFAIU FTDI, the state of CBUS pins is not supposed to change when the mode changes.

> Each pin can be independently set as an input or an output. If no new data is written to the device, the output pins will hold the last value written. Note that after setting any CBUS pin to GPIO mode; it will default to be an input pulled to VCCIO via internal resistors. When the bit mode is reset to normal operation (mode = 0x00), the CBUS GPIO pins will continue to hold their previous configuration and value. (To avoid having any pins continuing to drive out after a bit mode reset, set them to be inputs prior to performing the reset.)

Although this is document for the FT-X series, I'm not quite sure for older devices, I've not been able to find the documentation.

I guess I need to check the behavior with FT232R...

mani9876, 2020-07-27:
@eblot 
Thank you for your feedback. Am I able to help you? Maybe I can send you some log files, or debug files, to investigate that problem further?
I am running my program at the moment with the edited ftdi.py, and this works fine.

mgrela, 2022-08-29:
I have encountered this problem as well and figured out that the following solution works without any changes to the pyftdi code:

```
#!/usr/bin/env python3

import time

from pyftdi.ftdi import Ftdi
from pyftdi.eeprom import FtdiEeprom

device = 'ftdi://ftdi/1'

ftdi = Ftdi()
ftdi.open_from_url(url=device)

# Set mode manually
ftdi.set_bitmode(0, Ftdi.BitMode.CBUS)
# ^^^^^^^^^^^^^

ftdi.set_cbus_direction(0b1111,0b1111)

while True:
    print('1111')
    ftdi.set_cbus_gpio(0b1111)
    time.sleep(1)

    print('0000')
    ftdi.set_cbus_gpio(0b0000)
    time.sleep(1)
```

mgrela, 2022-08-29:
> This is weird because AFAIU FTDI, the state of CBUS pins is not supposed to change when the mode changes.
>
>
> Although this is document for the FT-X series, I'm not quite sure for older devices, I've [...]

I was thinking maybe it's a difference between the original FT232R and one of those knockoff chips. Do you know of a way to figure out whether the FT232R you have is genuine?

mani9876, 2022-09-21:
> I have encountered this problem as well and figured out that the following solution works without any changes to the pyftdi code:
>
> ```
> #!/usr/bin/env python3
>
> import time
>
> from pyftdi.ftdi import [...]

Hello!

Thank you for the reply, I've tried your fix, but unfortunately, it raises an error:

```
Traceback (most recent call last):
  File "/root/./final.py", line 245, in <module>
    ftdi.set_cbus_gpio(0b1011)
  File "/usr/lib/python3.9/site-packages/pyftdi/ftdi.py", line 1309, in set_cbus_gpio
    self.set_bitmode(value, Ftdi.BitMode.CBUS)
  File "/usr/lib/python3.9/site-packages/pyftdi/ftdi.py", line 1246, in set_bitmode
    if self._ctrl_transfer_out(Ftdi.SIO_REQ_SET_BITMODE, value):
  File "/usr/lib/python3.9/site-packages/pyftdi/ftdi.py", line 2069, in _ctrl_transfer_out
    raise FtdiError('UsbError: %s' % str(ex)) from None
pyftdi.ftdi.FtdiError: UsbError: [Errno 19] No such device (it may have been disconnected)
```

I changed my script to the following:
```
#set UART
port = pyftdi.serialext.serial_for_url(serial, baudrate=19200, timeout=10)
ftdi = port.ftdi
ftdi.set_bitmode(0, Ftdi.BitMode.CBUS) 
ftdi.set_cbus_direction(0b1111,0b1111)
```

Have you got any further ideas?

Thanks!
Manuel

mgrela, 2022-09-22:
---- On Wed, 21 Sep 2022 13:39:06 +0200  mani9876  wrote --- 

>

> I have encountered this problem as well and figured out that the following solution works without any changes to the pyftdi code:

> #!/usr/bin/env python3import timefrom pyftdi.ftdi import Ftdifrom pyftdi.eeprom import FtdiEepromdevice = 'ftdi://ftdi/1'ftdi = Ftdi()ftdi.open_from_url(url=device)# Set mode [...]

> Thank you for the reply, I've tried your fix, but unfortunately, it raises an error:

> Traceback (most recent call last):  File "/root/./final.py", line 245, in     ftdi.set_cbus_gpio(0b1011)  File "/usr/lib/python3.9/site-packages/pyftdi/ftdi.py", line 1309, in set_cbus_gpio    self.set_bitmode(value, Ftdi.BitMode.CBUS)  File "/usr/lib/python3.9/site-packages/pyftdi/ftdi.py", line 1246, in set_bitmode    if self._ctrl_transfer_out(Ftdi.SIO_REQ_SET_BITMODE, value):  File "/usr/lib/python3.9/site-packages/pyftdi/ftdi.py", line 2069, in _ctrl_transfer_out    raise FtdiError('UsbError: %s' % str(ex)) from Nonepyftdi.ftdi.FtdiError: UsbError: [Errno 19] No such device (it may have been disconnected)I changed my script to the following:

> #set UARTport = pyftdi.serialext.serial_for_url(serial, baudrate=19200, timeout=10)ftdi = port.ftdiftdi.set_bitmode(0, Ftdi.BitMode.CBUS) ftdi.set_cbus_direction(0b1111,0b1111)Have you got any further ideas?

I can't see this message on github. Can you reply there so that everyone can participate in the discussion? As for your particular problem I do not know whether you can use the normal serial mode and CBUS mode concurrently with pyftdi. They use entirely different system-level APIs. If you have the device open as a serial port than pyftdi might get an error code from the kernel. I think you should try to get *anything* working with open_from_url (maybe the URL in my example is not correct for you)?

--

enki

mani9876, 2022-09-25:
> ---- On Wed, 21 Sep 2022 13:39:06 +0200  mani9876  wrote ---
> I have encountered this problem as well and figured out that the following solution works without any changes to the pyftdi code: [...]

Hello!

Sorry, i deleted the post, at the error was my mistake ;-).
I just want to thank you for the hint, now everything works as expected, without changing the pyftdi.py file :-)!

Best regards
Manuel

mgrela, 2022-09-25:
---- On Sun, 25 Sep 2022 09:21:48 +0200  mani9876  wrote --- 
 > 
 > ---- On Wed, 21 Sep 2022 13:39:06 +0200  mani9876  wrote ---
 > I have encountered this problem as well and figured out that the following solution works without any changes to the pyftdi code: #!/usr/bin/env python3import timefrom pyftdi.ftdi import Ftdifrom pyftdi.eeprom import FtdiEepromdevice = 'ftdi://ftdi/1'ftdi = Ftdi()ftdi.open_from_url(url=device)# Set mode manuallyftdi.set_bitmode(0, Ftdi.BitMode.CBUS)# ^^^^^^^^^^^^^ftdi.set_cbus_direction(0b1111,0b1111)while True:    print('1111')    ftdi.set_cbus_gpio(0b1111)    time.sleep(1)    print('0000')    ftdi.set_cbus_gpio(0b0000)    time.sleep(1)Hello! Thank you for the reply, I've tried your fix, but unfortunately, it raises an error: Traceback (most recent call last):  File "/root/./final.py", line 245, in     ftdi.set_cbus_gpio(0b1011)  File "/usr/lib/python3.9/site-packages/pyftdi/ftdi.py", line 1309, in set_cbus_gpio    self.set_bitmode(value, Ftdi.BitMode.CBUS)  File "/usr/lib/python3.9/site-packages/pyftdi/ftdi.py", line 1246, in set_bitmode    if self._ctrl_transfer_out(Ftdi.SIO_REQ_SET_BITMODE, value):  File "/usr/lib/python3.9/site-packages/pyftdi/ftdi.py", line 2069, in _ctrl_transfer_out    raise FtdiError('UsbError: %s' % str(ex)) from Nonepyftdi.ftdi.FtdiError: UsbError: [Errno 19] No such device (it may have been disconnected)I changed my script to the following: #set UARTport = pyftdi.serialext.serial_for_url(serial, baudrate=19200, timeout=10)ftdi = port.ftdiftdi.set_bitmode(0, Ftdi.BitMode.CBUS) ftdi.set_cbus_direction(0b1111,0b1111)Have you got any further ideas?
 > I can't see this message on github. Can you reply there so that everyone can participate in the discussion? As for your particular problem I do not know whether you can use the normal serial mode and CBUS mode concurrently with pyftdi. They use entirely different system-level APIs. If you have the device open as a serial port than pyftdi might get an error code from the kernel. I think you should try to get anything working with open_from_url (maybe the URL in my example is not correct for you)?
 > …
 > -- enki
 > Hello!
 > Sorry, i deleted the post, at the error was my mistake ;-).
 > I just want to thank you for the hint, now everything works as expected, without changing the pyftdi.py file :-)!

Excellent, glad to hear it!

-- 
enki

enkiusz, 2022-10-27:
> > ---- On Wed, 21 Sep 2022 13:39:06 +0200  mani9876  wrote ---
[...]
> 
> Sorry, i deleted the post, at the error was my mistake ;-). I just want to thank you for the hint, now everything works as expected, without changing the pyftdi.py file :-)!
> 

Can you mark the issue as closed please?

closed 2022-10-27 by mani9876

FabianSchwartau, 2023-04-21:
This should be mentioned in the docs...