eblot/pyftdi #218: Retain GPO (PIN) status on re-configuration

mschuetterle, 2020-09-02:
Hi,

We are using PyFtdi with a dual FTDI (FT2232) chip by connecting port A in MPSSE mode:
```
port = GpioMpsseController()
/# Configure Bit13 as Output
port.configure('ftdi://ftdi:ft2232/1', direction=0x2000, frequency=10e6)
```

We have seen that after toggling Bit13 and closing the port `port.close()` the value got retained.
**But when re-running the SW and again configuring the port, the Bit13 got a reset!**

Even using `direction=0` to read Bit13 first, does not change the behavior. So, my assumption, this is an issue on the configuration of the Port into MPSSE mode.

Also the `initial` value does not help, as we just want to retain the previous value of Bit13 - which is unknown.

Is there any idea or other way to resolve this?

eblot, 2020-09-02:
I would say this is by design: whenever you close the port, you release the hardware resources, and revert them to the idle state. I guess you could hack into the code to prevent PyFtdi from performing a port reset on release.

Whenever the port is open, the interface is reset - because there is no other way to start from a known state - so I'm afraid re-opening always means resetting.

Why do you close the port if you need to keep using the device BTW?

mschuetterle, 2020-09-02:
Thank you for the feedback.

Just want to correct a misunderstanding:
The hardware resources are not been reverted to idle state when closing the port, but when opening (re-configuring) the next time. This was something unexpected to me.

Your last question is perfectly the right one. Let me help to clarify:
- Our SW could be considered a Daemon running in the background.
- The PIN13 is a RESET signal for other HW and will release them when our SW is started.
- Closing the port is just part of cleanup in our SW for example when shutting down the system.

Now, why I'm asking?
We are doing some test which require to frequently restart our SW (Daemon) and this is causing each time a reset cycle on the connected HW.
So, I tried to find a way that the PIN state from previous run will not be changed.

Thank you for your support

eblot, 2020-09-02:
Ok I got it.

A heavy duty solution would be to fork a daemon to manage FTDI, so that it stays alive even if the SW daemon is reset.

Unfortunately there is no way to know the current state of the FTDI HW, so not resetting it whenever the port is open would lead to unpredictable behaviour, but I guess you can hack you own copy of PyFtdi to discard all reset class and see if it helps, but I'm afraid it gonna lead to some weird behaviour one day or another.

mschuetterle, 2020-09-02:
Okay, again thank you for the feedback!

closed 2020-09-02 by mschuetterle

eblot, 2021-05-05:
There is now an option to preserve the output state of GPIO pin when the port is closed, see #219.

However, it does not address this initial requirement: the port will still be re-initialized on open.